OpenBCI/OpenBCI_GUI #977: Integrate brainflow MLModels

markopetrovic-neurotech, 2021-06-03:
Hello all,

I have recently taken up a project to integrate a BCI into outside applications to show concentration, stress, etc... After some research OpenBCI and Brainflow look like the best place to start. I was able to create a websocket server that sends raw data from the update() function in board.pde, however I am unsure of how to implement MLModels within the app.

Here is an example of one of the algorithms from the brainflow website.

` Pair<double[], double[]> bands = DataFilter.get_avg_band_powers (data, eeg_channels, sampling_rate, true);
        double[] feature_vector = ArrayUtils.addAll (bands.getLeft (), bands.getRight ());
        BrainFlowModelParams model_params = new BrainFlowModelParams (BrainFlowMetrics.CONCENTRATION.get_code (),
                BrainFlowClassifiers.REGRESSION.get_code ());
        MLModel concentration = new MLModel (model_params);
        concentration.prepare ();
        System.out.print ("Concentration: " + concentration.predict (feature_vector));
        concentration.release ();`

So far I have tried adding this to the same update() function. It works briefly however the main process crashes after a few seconds, I believe because it is interfering with the main render loop. I'm writing here now in hopes that someone can guide me in the right direction for how to implement this feature. Any help would be much appreciated.

retiutut, 2021-06-03:
This has just been completed and rolled out to the public!!! Just need to add documentation (plan to do in the next week).

You can track the progress and dig through the changes to the code/PR here:
https://github.com/OpenBCI/OpenBCI_GUI/issues/924

I hope that we can start to improve the Models to increase the accuracy of these metrics in the near future.

Closing this issue.

Thanks for your interest in this feature, and happy to share that it is already done!

Take Care,
RW

closed 2021-06-03 by retiutut

markopetrovic-neurotech, 2021-06-04:
Wow that's amazing! 

What kind of improvements need to made for the models?

retiutut, 2021-06-04:
> Wow that's amazing!
> 
> Is there a thread to brainstorm/track the improvement of these models?

I’m not really a data scientist, but we have our pick of a few classification algorithms and we could further fine tune the “models” to increase the accuracy of “relaxation” and “concentration” metrics.