OpenBCI/OpenBCI_GUI #978: Openbci gui crashes

CsimbokDavid, 2021-06-18:
## Problem

When im pressing start data stream, the whole gui just freezes and nothing happens.

## Expected 

To start streaming data, and don't freeze

## Operating System and Version
Windows

## GUI Version
v5.0.5

## Running standalone app
Downloaded app

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

## Console Log
Just crashes, can't check the log, when i clicked to show it before i pressed the start data stream, it showed 1 empty line at the start then crashed.
Paste any relevant text from the console window here

retiutut, 2021-06-23:
Please provide more information, including Windows OS version and Console Log.

Console Logs can be found in [USER]/Documents/OpenBCI_GUI/Console_Data.

CsimbokDavid, 2021-06-24:
windows 10, 
[Console_2021-06-22_11-21-51.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/6707164/Console_2021-06-22_11-21-51.txt)
[Console_2021-06-20_00-37-13.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/6707170/Console_2021-06-20_00-37-13.txt)

retiutut, 2021-06-24:
@CsimbokDavid The GUI needs permission to write data to file in your Documents folder. While it is able to create a console log file, something is preventing Java from creating a file writer to save data to file.

closed 2021-06-28 by CsimbokDavid