OpenBCI/OpenBCI_GUI #980: Cannot start session - Cyton purchased in Oct 2017

alexjamesmajor, 2021-06-26:
## Problem

I bought this board years ago and am now trying to set it up. I was able to successfully connect to the Cyton once and saw some voltage signals, but since then I cannot connect. I'm not sure what caused this change. I also have a Daisy board, but I am keeping that disconnected while I troubleshoot.

I have spent several hours scanning the OpenBCI documents for ideas, but am at the point where I would appreciate some guidance. Here is some console window text:

```
Console Log Started at Local Time: 2021-06-26_17-11-40
Screen Resolution: 1920 X 1080
Operating System: Windows
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
OpenBCI_GUI::Setup: GUI v5 Sample Data exists in Documents folder.
ControlP5 2.3.1 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
TopNav: Internet Connection Successful
Local Version: v5.0.5, Latest Version: v5.0.5
GUI is up to date!
OpenBCI_GUI::Control Panel: Playback history file not found.
OpenBCI_GUI::Setup: Setup is complete!
sourceList: click! 
OpenBCI_GUI: Channel count set to 8
ControlPanel: Found Cyton Dongle on COM port: COM5
cytonDongleList: click! 
[DEFAULT]: ControlPanel: Selected OpenBCI Port COM5
[DEFAULT]: Attempting to Start Session...
initButtonPressed: Calling initSystem() after next draw()
OpenBCI_GUI: Start session. Calling initSystem().

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

[ERROR]: ERROR: brainflow.BrainFlowError: Error in prepare_session:BOARD_NOT_READY_ERROR when initializing Brainflow board. Data will not stream.
brainflow.BrainFlowError: Error in prepare_session:BOARD_NOT_READY_ERROR
	at brainflow.BoardShim.prepare_session(BoardShim.java:598)
	at OpenBCI_GUI$BoardBrainFlow.initializeInternal(OpenBCI_GUI.java:2487)
	at OpenBCI_GUI$BoardCyton.initializeInternal(OpenBCI_GUI.java:3013)
	at OpenBCI_GUI$Board.initialize(OpenBCI_GUI.java:2018)
	at OpenBCI_GUI.initSystem(OpenBCI_GUI.java:691)
	at OpenBCI_GUI.systemInitSession(OpenBCI_GUI.java:1009)
	at OpenBCI_GUI.draw(OpenBCI_GUI.java:569)
	at processing.core.PApplet.handleDraw(PApplet.java:2475)
	at processing.opengl.PSurfaceJOGL$DrawListener.display(PSurfaceJOGL.java:866)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
	at jogamp.opengl.GLAutoDrawableBase$2.run(GLAutoDrawableBase.java:443)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at com.jogamp.newt.opengl.GLWindow.display(GLWindow.java:759)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:452)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
OpenBCI_GUI: Configuring Cyton Channel Count...
OpenBCI_GUI: Channel count set to 8
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
[DEFAULT]: Data stream is already stopped.
[ERROR]: Failed to initialize board. Please check that the board is on and has power. See Console Log for more details.
OpenBCI_GUI::Control Panel: Playback history file not found.

-------------------------------------------------
ControlPanel: Attempting to Auto-Connect to Cyton
-------------------------------------------------

ControlPanel: Found Cyton Dongle on COM port: COM5
[WARN]: Found a Cyton dongle, but could not connect to the board. Auto-Scanning now...
[ERROR]: Unable to connect to Cyton. Please check hardware and power source.
cytonDongleList: click! 
[DEFAULT]: ControlPanel: Selected OpenBCI Port COM5
ControlPanel: Found Cyton Dongle on COM port: COM5
cytonDongleList: click! 
[DEFAULT]: ControlPanel: Selected OpenBCI Port COM5

-------------------------------------------------
ControlPanel: Attempting to Auto-Connect to Cyton
-------------------------------------------------

ControlPanel: Found Cyton Dongle on COM port: COM5
[WARN]: Found a Cyton dongle, but could not connect to the board. Auto-Scanning now...
[ERROR]: Unable to connect to Cyton. Please check hardware and power source.
ConsoleLog: Copying console log to clipboard!

```

>>>>>>>
Also, I have tried to update the firmware with little success, by trying to flash on DefaultBoard. Here is the error message from Arduino (1.8.15) when I try to upload DefaultBoard:
>>>>>>>

Arduino: 1.8.15 (Windows 10), Board: "OpenBCI 32"

Sketch uses 84916 bytes (69%) of program storage space. Maximum is 122880 bytes.

Global variables use 10564 bytes (32%) of dynamic memory, leaving 22204 bytes for local variables. Maximum is 32768 bytes.

An error occurred while uploading the sketch

    Copyright: (C) 2011-2015 Serge Vakulenko
                   2016-2017 Majenko Technologies

No target found.

Programmer for Microchip PIC32 microcontrollers, Version 2.1.46

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

## Expected 

Any next steps in troubleshooting would be greatly appreciated!

## Operating System and Version

Windows 10

## GUI Version

5.05

## Running standalone app

Standalone (haven't tried Processing yet)

## Type of OpenBCI Board

Currently Cyton (I will try Cyton + Daisy after I can connect)

## Are you using a WiFi Shield?

Haven't purchased this yet.

## Console Log

see above

retiutut, 2021-06-30:
Hey @alexjamesmajor what happened with this? What resolved the issue?

retiutut, 2021-06-30:
Original Post:

Problem

I bought this board years ago and am now trying to set it up. I was able to successfully connect to the Cyton once and saw some voltage signals, but since then I cannot connect. I'm not sure what caused this change. I also have a Daisy board, but I am keeping that disconnected while I troubleshoot.

I have spent several hours scanning the OpenBCI documents for ideas, but am at the point where I would appreciate some guidance. Here is some console window text:

Console Log Started at Local Time: 2021-06-26_17-11-40
Screen Resolution: 1920 X 1080
Operating System: Windows
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
OpenBCI_GUI::Setup: GUI v5 Sample Data exists in Documents folder.
ControlP5 2.3.1 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
TopNav: Internet Connection Successful
Local Version: v5.0.5, Latest Version: v5.0.5
GUI is up to date!
OpenBCI_GUI::Control Panel: Playback history file not found.
OpenBCI_GUI::Setup: Setup is complete!
sourceList: click! 
OpenBCI_GUI: Channel count set to 8
ControlPanel: Found Cyton Dongle on COM port: COM5
cytonDongleList: click! 
[DEFAULT]: ControlPanel: Selected OpenBCI Port COM5
[DEFAULT]: Attempting to Start Session...
initButtonPressed: Calling initSystem() after next draw()
OpenBCI_GUI: Start session. Calling initSystem().

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

[ERROR]: ERROR: brainflow.BrainFlowError: Error in prepare_session:BOARD_NOT_READY_ERROR when initializing Brainflow board. Data will not stream.
brainflow.BrainFlowError: Error in prepare_session:BOARD_NOT_READY_ERROR
	at brainflow.BoardShim.prepare_session(BoardShim.java:598)
	at OpenBCI_GUI$BoardBrainFlow.initializeInternal(OpenBCI_GUI.java:2487)
	at OpenBCI_GUI$BoardCyton.initializeInternal(OpenBCI_GUI.java:3013)
	at OpenBCI_GUI$Board.initialize(OpenBCI_GUI.java:2018)
	at OpenBCI_GUI.initSystem(OpenBCI_GUI.java:691)
	at OpenBCI_GUI.systemInitSession(OpenBCI_GUI.java:1009)
	at OpenBCI_GUI.draw(OpenBCI_GUI.java:569)
	at processing.core.PApplet.handleDraw(PApplet.java:2475)
	at processing.opengl.PSurfaceJOGL$DrawListener.display(PSurfaceJOGL.java:866)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
	at jogamp.opengl.GLAutoDrawableBase$2.run(GLAutoDrawableBase.java:443)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at com.jogamp.newt.opengl.GLWindow.display(GLWindow.java:759)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:452)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
OpenBCI_GUI: Configuring Cyton Channel Count...
OpenBCI_GUI: Channel count set to 8
openBCI_GUI: haltSystem: Halting system for reconfiguration of settings...
[DEFAULT]: Data stream is already stopped.
[ERROR]: Failed to initialize board. Please check that the board is on and has power. See Console Log for more details.
OpenBCI_GUI::Control Panel: Playback history file not found.

-------------------------------------------------
ControlPanel: Attempting to Auto-Connect to Cyton
-------------------------------------------------

ControlPanel: Found Cyton Dongle on COM port: COM5
[WARN]: Found a Cyton dongle, but could not connect to the board. Auto-Scanning now...
[ERROR]: Unable to connect to Cyton. Please check hardware and power source.
cytonDongleList: click! 
[DEFAULT]: ControlPanel: Selected OpenBCI Port COM5
ControlPanel: Found Cyton Dongle on COM port: COM5
cytonDongleList: click! 
[DEFAULT]: ControlPanel: Selected OpenBCI Port COM5

-------------------------------------------------
ControlPanel: Attempting to Auto-Connect to Cyton
-------------------------------------------------

ControlPanel: Found Cyton Dongle on COM port: COM5
[WARN]: Found a Cyton dongle, but could not connect to the board. Auto-Scanning now...
[ERROR]: Unable to connect to Cyton. Please check hardware and power source.
ConsoleLog: Copying console log to clipboard!

Also, I have tried to update the firmware with little success, by trying to flash on DefaultBoard. Here is the error message from Arduino (1.8.15) when I try to upload DefaultBoard:

Arduino: 1.8.15 (Windows 10), Board: "OpenBCI 32"

Sketch uses 84916 bytes (69%) of program storage space. Maximum is 122880 bytes.

Global variables use 10564 bytes (32%) of dynamic memory, leaving 22204 bytes for local variables. Maximum is 32768 bytes.

An error occurred while uploading the sketch

Copyright: (C) 2011-2015 Serge Vakulenko
               2016-2017 Majenko Technologies
No target found.

Programmer for Microchip PIC32 microcontrollers, Version 2.1.46

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Expected

Any next steps in troubleshooting would be greatly appreciated!

Operating System and Version

Windows 10

GUI Version

5.05

Running standalone app

Standalone (haven't tried Processing yet)

Type of OpenBCI Board

Currently Cyton (I will try Cyton + Daisy after I can connect)

Are you using a WiFi Shield?

Haven't purchased this yet.

Console Log

see above

closed 2021-06-30 by retiutut

alexjamesmajor, 2021-07-01:
Hi RW,

Thanks for following up. I removed the post so that I could ask the question on the OpenBCI forum, which appears to be more active. However, before I reposted, I decided to try all the steps again, which fixed my issue. I am not sure which step helped fix the problem exactly, but if I were to guess I would say it was because I neglected to call radio.flashnonvoltalememoryetc(); during either the Host or Device firmware upgrade.

For completeness, here are all the steps I took:

(1) Device firmware upgrade
(2) Host firmware upgrade
(3) Load Defaultboard/pic32 step

I also upgraded USB ports to FTDI, tried everything with Proccessing instead of standalone GUI and tried different USB ports.

Possibly an important step to include in the documentation is restarting one's computer - this seemed to help as well.

In general I would agree with other posters that the documentation (specifically firmware upgrades) is a bit difficult to navigate. Especially for people without much electrical eng or Ardruino experience. One helpful piece of info that would be helpful is to add _how to confirm that firmware upgrades worked_. After I was successful, I could see the unique flashing lights on the device and the "SUCCESS" message, but during a previous unsuccessful attempt at holding the pins against the solder etc, I had no clue if it was successful or not because I didn't know what to look for. All in all, it was rewarding to get it to work and now I have 99 unused 0.1 nF capacitors to play with haha.

Thanks for your time, I look forward to testing out this device more! I think the OpenBCI initiative is a great way for hobbyists to enter the space. Hopefully I'll see your crew at an upcoming SfN. I'm a neuro postdoc myself and will be applying to try out the Galea once I have more experience with OpenBCI!

Best,

Alex