OpenBCI/OpenBCI_GUI #987: OpenBCI GUI v5.0.5 shows only a blank window upon startup on my MacBook M1

ScottThomasMiller, 2021-08-12:
## Problem

When I open the GUI, it displays only a blank window.  

[image]

## Expected 

I expect to see the usual startup window.

[image]

## Operating System and Version

macOS Big Sur 11.5.1 (20G80) on MacBook M1

## GUI Version

5.0.5

## Running standalone app

downloaded app 

## Type of OpenBCI Board

Cyton+Daisy

## Are you using a WiFi Shield?

No

## Console Log

N/A

ScottThomasMiller, 2021-08-12:
Prior versions work fine.  I dropped back to v5.0.4, which seems to run without any issues.

ScottThomasMiller, 2021-08-12:
Sorry about the confusion.  The non-working version is 5.0.6.  The working version is 5.0.5.

retiutut, 2021-08-13:
@ScottThomasMiller Thank you for making this issue. 

Can you try and locate the Console Log file for 5.0.6 in `/USER/Documents/OpenBCI_GUI/Console_Data` and share it here?

If the window opens, I have a feeling there might also be a Console Log and specific info on what could be causing the issue.

ScottThomasMiller, 2021-08-13:
It does not produce a console log.  I am looking in /Users/scottmiller/Documents/OpenBCI_GUI/Console_Data.

retiutut, 2021-08-16:
I was able to run 5.0.6 on a 2014 Mac Mini running Big Sur 11.5.2. Tagging this issue as hardware related.

Also tagging an upstream issue attempting to be tackled by the Processing team in Processing 4. https://github.com/processing/processing4/issues/128

retiutut, 2021-08-16:
@ScottThomasMiller Would you be willing to help debug this further by running the GUI from Processing?

It's interesting that we are running Processing 3, and 5.0.5 **does work**.....

Maybe it broke by implementing the following fix https://github.com/OpenBCI/OpenBCI_GUI/issues/968 which sets pixel density...

If you are available to help I can provide minimal instructions on how to test this.

ScottThomasMiller, 2021-08-16:
Absolutely!  Are these the instructions: https://github.com/openbci-archive/Docs/blob/master/Deprecated%20Docs/Run%20The%20OpenBI%20GUI%20In%20Processing.md

retiutut, 2021-08-16:
Ouch. Those are the old docs. Here is the new link: https://docs.openbci.com/Software/OpenBCISoftware/GUIDocs/#running-the-openbci-gui-from-the-processing-ide

Basically, install "processing-java" using the Processing IDE on Mac. Clone the OpenBCI GUI repo. Copy the `libraries` to the Processing `libraries` folder in `Documents`.

Then, I'd like to try removing the `pixelDensity()` function in OpenBCI_GUI.pde.

ScottThomasMiller, 2021-08-17:
Without editing the code to comment-out that call, I tried to run the app from Processing, but I get an error:

[image]

retiutut, 2021-08-17:
@ScottThomasMiller Let's stick with Processing 3 for now. Migrating to Processing 4 is another separate task.

ScottThomasMiller, 2021-08-17:
My bad.  Fixing...

ScottThomasMiller, 2021-08-17:
Good call:

[image]

retiutut, 2021-08-17:
@ScottThomasMiller Sweet... remove or comment it out and try again.

ScottThomasMiller, 2021-08-17:
Bingo!

[image]

retiutut, 2021-08-17:
@ScottThomasMiller Next, can you move that whole code block from `setup()` into `settings()`?

Move this stuff to the method above it
```
    StringBuilder sb_res = new StringBuilder("Screen Resolution: ");
    sb_res.append(displayWidth);
    sb_res.append(" X ");
    sb_res.append(displayHeight);
    //Account for high-dpi displays on Mac, Windows, and Linux Machines Fixes #968
    pixelDensity(displayDensity());
    StringBuilder sb_dpi = new StringBuilder("High-DPI Screen Detected: ");
    sb_dpi.append(displayDensity() == 2);
```

ScottThomasMiller, 2021-08-17:
Does this look correct?

void settings() {
    //LINUX GFX FIX #816
    System.setProperty("jogl.disable.openglcore", "false");

    win_w = 1024;
    win_h = 768;

    // If less than 1366x768, set smaller minimum GUI size
    // Nov 2020 - Accomodate as low as 1024 X 640
    if (displayWidth <= 1366 || displayHeight <= 768) {
        win_w = 980;
        win_h = 580;
    }
    size(win_w, win_h, P2D);
    
    StringBuilder sb_res = new StringBuilder("Screen Resolution: ");
    sb_res.append(displayWidth);
    sb_res.append(" X ");
    sb_res.append(displayHeight);
    //Account for high-dpi displays on Mac, Windows, and Linux Machines Fixes #968
    //pixelDensity(displayDensity());
    StringBuilder sb_dpi = new StringBuilder("High-DPI Screen Detected: ");
    sb_dpi.append(displayDensity() == 2);
}

void setup() {
    frameRate(120);

    copyPaste = new CopyPaste();

retiutut, 2021-08-17:
Yes. Give that a go.

ScottThomasMiller, 2021-08-17:
[image]

retiutut, 2021-08-17:
Comment out lines 384 and 385. The variables are not accessible since you moved it.

ScottThomasMiller, 2021-08-17:
That works.  The GUI opens once again.

retiutut, 2021-08-17:
@ScottThomasMiller Can you share the console log output now after starting the GUI successfully? Move those println()s you just changed to the end of the `settings()` method so I can see the DPI and screen resolution you are using as detected by Java.

ScottThomasMiller, 2021-08-17:
Console log:

```
Console Log Started at Local Time: 2021-08-16_20-36-59
Operating System: Mac
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
OpenBCI_GUI::Setup: GUI v5 Sample Data exists in Documents folder.
ControlP5 2.3.1 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
TopNav: Internet Connection Successful
Local Version: v5.0.6, Latest Version: v5.0.6
GUI is up to date!
OpenBCI_GUI: AuditoryFeedback: Loading Audio...
OpenBCI_GUI: AuditoryFeedback: Done Loading Audio!
OpenBCI_GUI::Control Panel: Playback history file not found.
OpenBCI_GUI::Setup: Setup is complete!
ConsoleLog: Copying console log to clipboard!

```

ScottThomasMiller, 2021-08-17:
Screen Resolution: 1504 X 846
High-DPI Screen Detected: true

retiutut, 2021-08-17:
> Screen Resolution: 1504 X 846
> High-DPI Screen Detected: true

That's a weird resolution. What size MacBook is this? Can you also open up the GUI in `Synthetic` data mode, press `Start Data Stream`, let some data fill up the screen and share another screenshot?

ScottThomasMiller, 2021-08-17:
It's a 14-inch MacBook M1, but it's closed.  I am using a docking station and a ViewSonic monitor in Scaled mode with large fonts, because my eyes are no longer young.

Here's your synthetic-mode screenshot:

[image]

retiutut, 2021-08-17:
Is this a MacBook Air or MacBook pro? Can you do this same test and share a screenshot using the MacBook's display?

This is the last test.

ScottThomasMiller, 2021-08-17:
Pro.  From the clamshell:

[image]

retiutut, 2021-08-17:
I think you have adjusted the display to a lower resolution, which is fine. It's great to see these screenshots to confirm that two critical pieces of software are working:

1. The GUI using Processing 3
2. BrainFlow Java Binding on M1 hardware (since a synthetic board is being used in this case)

Using Processing IDE, you are free to "Export Application" now to have a version you can use for the next week or so until 5.0.7 is released towards the end of the month.

**Thank you so much for your time and effort to help on this matter!!!** This was much cheaper than going out to buy a new Mac just to confirm an easy fix.

💯💯💯💯💯💯💯💯💯💯

ScottThomasMiller, 2021-08-17:
My pleasure.  Thanks for the speedy fix.

Daisykh, 2021-08-19:
I believe I ran into the same issue (#990) using a 2019 intel macbook pro on Catalina v10.15.17. I went through the same steps and here are the results:

1. Running via processing3, no changes made and results in the blank screen. Here's the console output:
[image]

2. Commented out `pixelDensity` and it now opens! Here's the console log output that includes the screen size and High-DPI detection of my macbook:
[image]

3. I also tried moving `pixelDensity(displayDensity())` into the `settings` function and it produced the same results as commenting it out entirely. Here's a screen grab:
[image]

PR #995 (2021-09-03): Fix highDPI code breaking GUI on some Macs
Fixes #987
Fixes #990
commit ed05ac71 (2021-09-03): Fix highDPI code breaking GUI on some Macs
Fixes #987
Fixes #990
closed 2021-09-06 by retiutut