OpenBCI/OpenBCI_GUI #989: When streaming LSL data, every fifth timestamp is too quick

ScottThomasMiller, 2021-08-17:
## Problem

I am pulling LSL samples into my custom iOS app, from the OpenBCI GUI in LSL time series streaming mode.  I am using 16 channels, so I expect my sampling rate to be 125Hz (0.008s.)  Regardless of whether I pull samples one-at-a-time or in chunks, every fifth timestamp seems to arrive too quickly, <<0.008s.  For example:

[image]

## Expected 

All timestamp intervals should reflect the sampling rate of 125Hz = 0.008s

## Operating System and Version

macOS Big Sur 11.5.1 on MacBook Pro M1

## GUI Version

5.0.5

## Running standalone app

standalone

## Type of OpenBCI Board

Cyton+Daisy

## Are you using a WiFi Shield?

No

## Console Log

I can provide a console log upon request.

ScottThomasMiller, 2021-08-18:
Chadwick @ LSL provided a detailed explanation of this behavior (he's been absolutely awesome), but I want to hear it from you all as well.  

Additionally, when I drop down to 8 channels, the sample rate doubles to 250Hz, and the period of the "quick" samples doubles from every 5 to every 10 samples.

What I need to know is: are the timestamps of the "quick" samples the actual LSL timestamps?

retiutut, 2021-08-18:
@ScottThomasMiller Thank you for digging into the LSL streaming of the TimeSeries data type. I've tagged this issue on #971 so as to consider this additional information. 

I will likely have more questions soon. 

Seems like this could be replicated and investigated using a Python sketch. 🤔

retiutut, 2021-09-18:
LSL calculates these timestamps. I'm not sure how it's something we could fix, unless we need to update the LSL Java dependency that the GUI uses.

ScottThomasMiller, 2021-09-18:
It's not something OpenBCI can fix.  I understand the fundamental issue much better now.  Thanks for looking into it.  Should we close this issue?

retiutut, 2021-09-18:
@ScottThomasMiller Can you explain what's going on with this to me briefly? I'd still like to understand.

ScottThomasMiller, 2021-09-18:
The root cause of this particular issue, if I remember correctly, is described by Chadwick here:

https://labstreaminglayer.slack.com/archives/C87GN1S4E/p1628990241050700?thread_ts=1628811549.047400&cid=C87GN1S4E

From that thread:

"The OpenBCI GUI app creates a stream with a nominal rate of 125 Hz. It then pushes a chunk of samples -- 5 at a time. LSL tags the last sample with the current lsl_local_clock time, then decrements the previous sample times in the chunk by 1/nominal_rate (=8 msec)."

--Scott

closed 2021-10-02 by retiutut