OpenBCI/OpenBCI_GUI #990: OpenBCI GUI v5.06 opens blank screen - macOS Catalina

Daisykh, 2021-08-19:
## Problem

When updated from 5.05 to 5.06, the OpenBCI GUI fails to open and displays a blank grey screen. v5.05 opens as expected with full functionality to connect to a Cyton board. I have tried 3 times to update and each time have had the same issue with 5.06 causing me to uninstall and reinstall 5.05. Image of GUI with version number below:
[image]

## Expected 

Opening the app should display the GUI as it does with 5.05. Image of previous version working below:
[image]

## Operating System and Version

macOS Catalina, v10.15.7
Macbook Pro 2019

## GUI Version

5.06
## Running standalone app

Downloaded GUI app
## Type of OpenBCI Board

Cyton board

## Are you using a WiFi Shield?

No

## Console Log

n/a, console log doesn't appear

retiutut, 2021-08-19:
https://github.com/OpenBCI/OpenBCI_GUI/issues/987

This is likely a duplicate issue that will have a fix in the next version of the GUI. To confirm, you could follow the same steps as in the other thread.

Daisykh, 2021-08-19:
Ahh! Yes, looks like the same issue even though I'm on an intel macbook pro and not M1! Here's the results:

1. Running via processing3, no changes made and results in the blank screen. Here's the console output:
[image]

2. Commented out `pixelDensity` and it now opens! Here's the console log output that includes the screen size and High-DPI detection of my macbook:
[image]

3. I also tried moving `pixelDensity(displayDensity())` into the `settings` function and it produced the same results as commenting it out entirely. Here's a screen grab:
[image]

This issue can be closed as a duplicate and I can relay my findings in #987 if it would be helpful.

retiutut, 2021-08-19:
Since this happened on different hardware and OS version, I'm going to keep this open until it's fixed.

Thank you for making an issue and following the template!

PR #995 (2021-09-03): Fix highDPI code breaking GUI on some Macs
Fixes #987
Fixes #990
commit ed05ac71 (2021-09-03): Fix highDPI code breaking GUI on some Macs
Fixes #987
Fixes #990
closed 2021-09-06 by retiutut