OpenBCI/OpenBCI_GUI #991: Unable to connect to Cyton.

zoot-io, 2021-08-19:
## Problem

When I try to start session, I am unable to connect to the Cyton board. 

[image]

 Usually, there's never any trouble at this stage. Two days ago without any updates to my computer, it stopped working. Superficially, both the dongle and the Cyton board are ok. The lights come on, but the suggested fix in the little bottom tab keeps suggesting a hardware issue: "Unable to connect to Cyton. Please check hardware and power source."

## Expected 

I expect to connect to the Cyton board so that I can start a session and collect data. 

## Operating System and Version

macOS Big Sur 11.4 on MacBook M1

## GUI Version

5.0.5

## Running standalone app

Downloaded app

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

## Console Log

```
Console Log Started at Local Time: 2021-08-19_16-57-15
Screen Resolution: 1440 X 900
Operating System: Mac
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://openbci.github.io/Documentation/docs/06Software/01-OpenBCISoftware/GUIDocs
OpenBCI_GUI::Setup: GUI v5 Sample Data exists in Documents folder.
ControlP5 2.3.1 infos, comments, questions at http://www.sojamo.de/libraries/controlP5
TopNav: Internet Connection Successful
Local Version: v5.0.5, Latest Version: v5.0.6
GUI needs to be updated. Download at https://github.com/OpenBCI/OpenBCI_GUI/releases/latest
OpenBCI_GUI::Control Panel: Playback history file not found.
OpenBCI_GUI::Setup: Setup is complete!
sourceList: click! 
OpenBCI_GUI: Channel count set to 8
ControlPanel: Found Cyton Dongle on COM port: /dev/cu.usbserial-DM0258P6
cytonDongleList: click! 
[DEFAULT]: ControlPanel: Selected OpenBCI Port /dev/cu.usbserial-DM0258P6

-------------------------------------------------
ControlPanel: Attempting to Auto-Connect to Cyton
-------------------------------------------------

ControlPanel: Found Cyton Dongle on COM port: /dev/cu.usbserial-DM0258P6
CytonAutoConnect: Error reading from Serial/COM port
[WARN]: Found a Cyton dongle, but could not connect to the board. Auto-Scanning now...
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
CytonAutoConnect: Error reading from Serial/COM port
[ERROR]: Unable to connect to Cyton. Please check hardware and power source.
ConsoleLog: Previous line copied to clipboard.
ConsoleLog: Copying console log to clipboard!

```

retiutut, 2021-08-23:
Something is preventing the GUI from accessing the dongle in a meaningful way...

zoot-io, 2021-08-24:
You're exactly right a few battles later I realized that the dongle had a switch, thanks!

Returned to close the issue.

closed 2021-08-24 by zoot-io