OpenBCI/OpenBCI_GUI #993: Problem with SD card Unsupported file, GUI 5.0,6

FabianGonzalez-ETH, 2021-08-23:
## Problem

I am saving data on the SD card using a Cyton board. The problem is that the GUI is saying "Trying to load an unsupported file for playback! Please try a valid file".
I already changed the SD card. I am currently using a Transcend Premium, 8Gb, UHS 1. 
I installed the latest version of the openBCI firmware 
I think the data is not well saved (I am sending an example of the data
[OBCI_07.TXT](https://github.com/OpenBCI/OpenBCI_GUI/files/7032779/OBCI_07.TXT)
[OBCI_08.TXT](https://github.com/OpenBCI/OpenBCI_GUI/files/7032781/OBCI_08.TXT)
)
I don't know if it is a problem with the GUI, the cyton or the SD card.

## Expected 

Reading the SD card without a problem and acquiring the data from it.

## Operating System and Version

Windows

## GUI Version

v5.0.6

## Running standalone app

Downloaded app

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

## Console Log

Paste any relevant text from the console window here

retiutut, 2021-09-16:
I am also getting this error with the file you shared on 5.0.5. Unfortunately, I don't think I have an extra SD card to test. 

**Will continue to look into this.** 

GUI v5 was tested with Cyton SD card files, and after the fix I don't believe anything has changed.

FabianGonzalez-ETH, 2021-09-24:
I changed the brand of the SD card, I am using now a sandisk ultra. After changing the SD  card, the files are not corrupted

closed 2021-09-24 by FabianGonzalez-ETH