OpenBCI/OpenBCI_GUI #1003: OpenBCI GUI v5.07 won't start in windows 10

LopezValdes, 2021-09-30:
## Problem

The release v5.07 from the 29 of September closes immediatly. OpenBCI_GUI.exe will open for half a second, showing a blank window, and then close immediately.

## Expected 

I xpected the GUI interface to load up for connection of my device

## Operating System and Version

Window 10 pro 64-bi version 20H2

## GUI Version

v5.07

## Running standalone app

Stand alone app

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

## Console Log

Paste any relevant text from the console window here

LopezValdes, 2021-09-30:
Just tried the previous release, that works as expected, so there is definitely an issue in the newest release. I hope that helps!

wrstscrnnm6, 2021-10-01:
Here is the error log for that crash. It had the same issue with the executable and in the processing IDE.

```
Operating System: Windows
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://docs.openbci.com/Software/OpenBCISoftware/GUIDocs/
OpenBCI_GUI::Setup: GUI v5 Sample Data exists in Documents folder.
OpenBCI_GUI::Settings: Creating new GUI-wide Settings file.
java.io.FileNotFoundException: C:\Users\XXXXXX\Documents\OpenBCI_GUI\Settings\GuiWideSettings.json (The system cannot find the path specified)
	at java.io.FileOutputStream.open0(Native Method)
	at java.io.FileOutputStream.open(FileOutputStream.java:270)
	at java.io.FileOutputStream.<init>(FileOutputStream.java:213)
	at java.io.FileOutputStream.<init>(FileOutputStream.java:101)
	at java.io.FileWriter.<init>(FileWriter.java:63)
	at OpenBCI_GUI$GuiSettings.saveToFile(OpenBCI_GUI.java:11804)
	at OpenBCI_GUI$GuiSettings.<init>(OpenBCI_GUI.java:11772)
	at OpenBCI_GUI.setup(OpenBCI_GUI.java:521)
	at processing.core.PApplet.handleDraw(PApplet.java:2425)
	at processing.opengl.PSurfaceJOGL$DrawListener.display(PSurfaceJOGL.java:866)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
	at jogamp.opengl.GLAutoDrawableBase$2.run(GLAutoDrawableBase.java:443)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at com.jogamp.newt.opengl.GLWindow.display(GLWindow.java:759)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:452)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
java.lang.NullPointerException
	at OpenBCI_GUI.output(OpenBCI_GUI.java:10436)
	at OpenBCI_GUI.outputWarn(OpenBCI_GUI.java:10452)
	at OpenBCI_GUI$GuiSettings.saveToFile(OpenBCI_GUI.java:11811)
	at OpenBCI_GUI$GuiSettings.<init>(OpenBCI_GUI.java:11772)
	at OpenBCI_GUI.setup(OpenBCI_GUI.java:521)
	at processing.core.PApplet.handleDraw(PApplet.java:2425)
	at processing.opengl.PSurfaceJOGL$DrawListener.display(PSurfaceJOGL.java:866)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
	at jogamp.opengl.GLAutoDrawableBase$2.run(GLAutoDrawableBase.java:443)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at com.jogamp.newt.opengl.GLWindow.display(GLWindow.java:759)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:452)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
handleDraw() called before finishing
```

There doesn't seem to be a settings folder in the directory nor the referenced `GuiWideSettings.json` file.

retiutut, 2021-10-01:
@wrstscrnnm6 Did you change the user name here before posting?

```
 C:\Users\XXXXXX\Documents\OpenBCI_GUI\Settings\GuiWideSettings.json
 ```

wrstscrnnm6, 2021-10-01:
>Did you change the user name here before posting?
yeah I changed it for the post. 
It was correctly aimed at my Documents folder.

retiutut, 2021-10-01:
There is enough information here to replicate (I think)! Will try to develop a fix by EOB today.

retiutut, 2021-10-01:
https://stackoverflow.com/questions/8668905/directory-does-not-exist-with-filewriter

retiutut, 2021-10-01:
[image]

Replicated!!! 🔥🔥🔥

PR #1005 (2021-10-02): Hot Fix NullPointer error related to missing folder and GUI-wide settings file
Fixes #1003 #1004
closed 2021-10-02 by retiutut