OpenBCI/OpenBCI_GUI #1004: OpenBCI_GUI – blank screen (Macbook)

0rang3-blu3, 2021-10-01:
## OpenBCI_GUI screen is blank (not launching) 

I'm trying to launch OpenBCI through Processing, but it's not coming up. I'm getting a blank screen. I'm receiving a "NullPointerException" note in Processing.

I also tried to launch the standalone app, but that's also not launching.

## Expected 

Ability to launch OpenBCI_GUI

## Operating System and Version

macOS

## GUI Version

OpenBCI_GUI v5.0.7

## Running standalone app

Running from Processing

## Are you using a WiFi Shield?

No

retiutut, 2021-10-01:
Please post the full console log file here.

0rang3-blu3, 2021-10-01:
Console Log Started at Local Time: 2021-10-01_09-49-03
Screen Resolution: 1680 X 1050
High-DPI Screen Detected: true
Operating System: Mac
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://docs.openbci.com/Software/OpenBCISoftware/GUIDocs/
OpenBCI_GUI::Setup: GUI v5 Sample Data exists in Documents folder.
OpenBCI_GUI::Settings: Creating new GUI-wide Settings file.
java.io.FileNotFoundException: /Users/username_*changed_for_post/Documents/OpenBCI_GUI/Settings/GuiWideSettings.json (No such file or directory)
	at java.base/java.io.FileOutputStream.open0(Native Method)
	at java.base/java.io.FileOutputStream.open(FileOutputStream.java:298)
	at java.base/java.io.FileOutputStream.<init>(FileOutputStream.java:237)
	at java.base/java.io.FileOutputStream.<init>(FileOutputStream.java:126)
	at java.base/java.io.FileWriter.<init>(FileWriter.java:66)
	at OpenBCI_GUI$GuiSettings.saveToFile(OpenBCI_GUI.java:11803)
	at OpenBCI_GUI$GuiSettings.<init>(OpenBCI_GUI.java:11771)
	at OpenBCI_GUI.setup(OpenBCI_GUI.java:522)
	at processing.core.PApplet.handleDraw(PApplet.java:2159)
	at processing.opengl.PSurfaceJOGL$DrawListener.display(PSurfaceJOGL.java:943)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
	at jogamp.opengl.GLAutoDrawableBase$2.run(GLAutoDrawableBase.java:443)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at com.jogamp.newt.opengl.GLWindow.display(GLWindow.java:782)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:81)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:453)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
	at java.base/java.util.TimerThread.mainLoop(Timer.java:556)
	at java.base/java.util.TimerThread.run(Timer.java:506)
handleDraw() called before finishing
NullPointerException

retiutut, 2021-10-01:
https://github.com/OpenBCI/OpenBCI_GUI/issues/1003

closed 2021-10-02 by retiutut