OpenBCI/OpenBCI_GUI #1014: SD Card Hex to Standard Text File Conversion script

retiutut, 2021-11-27:
## **Is your feature request related to a problem? Please describe.**
Some users would appreciate an SD card Hex to human-readable text file for Cyton created files.

Maybe the files should look like the OpenBCI CSV files exactly, mimicking the header information. Something in the script would also need to determine if a Cyton or Cyton Daisy was used.

## **Describe the solution you'd like**
Do this in Python. Shouldn't be that complicated or difficult to run.

## **Describe alternatives you've considered**
Current GUI reads straight from Hex values. Though, some users would like to maybe have a human-readable CSV file that can be used in the GUI or with another program. 

There used to be a conversion script in GUI 4.2.0, maybe we should not point users to this, since the data CSV format changed from GUI v4 to GUI v5

## **Additional context**
No other additional information.

retiutut, 2023-01-05:
https://openbci.com/forum/index.php?p=/discussion/3497/sdk-commands-to-get-the-data-files-from-sd-card-to-the-computer#latest

Another request for this feature. Upping the priority.

Shyamal-Dharia, 2023-08-30:
Hi, if possible I wanted to request you to prioritize this issue. 
Thanks

wjcroft, 2024-11-07:
So to summarize the current November 2024 status. User [roflecopter](https://github.com/roflecopter) has created a Python conversion script, done in January 2024.

https://github.com/roflecopter/openbci-session
https://github.com/roflecopter/openbci-session/blob/main/sd_convert.py

And user @Shyamal-Dharia has a GUI / Processing PR that will add a sd file convert feature to the GUI. Also completed and tested January 2024.

https://github.com/OpenBCI/OpenBCI_GUI/pull/1207

I believe both these conversion features should be made available to Cyton users. What is the next step? @retiutut @philippitts 

Regards,

closed 2025-04-01 by retiutut