OpenBCI/OpenBCI_GUI #1024: Remove date from prebuilt app files

hybras, 2021-12-21:
## Problem

[The released apps contain dates in their filenames](https://github.com/OpenBCI/OpenBCI_GUI/releases/tag/v5.0.9). This makes it slightly harder for automation to find the right file. Git tags / github releases already have dates, and you are using the dates alongside semantic version.

## Expected 

The filenames contain a subset of {app name, version, os, cpu arch}

retiutut, 2022-01-07:
Use startsWith and endsWith in your given language to resolve this. This decision was made by a very experienced software engineer at the architect level.

closed 2022-01-07 by retiutut

Andrey1994, 2022-01-12:
Hi, executables are distributed not only over github releases, and having dates or commit ids makes it easier to track the exact version and more obvious than adding smth to "about" window