OpenBCI/OpenBCI_GUI #1025: The type Serial is ambiguous.

cfausn, 2022-01-21:
## Problem

Issue starting the OpenBCI GUI with the latest version of Processing v4.0b3. When pressing "start," the following error occurs. **I resolved the issue by downgrading Processing to v3.5.4, but wanted to make sure ya'll were aware of the issue.**

## Expected 

Should start the OpenBCI GUI on pressing start.

## Operating System and Version

Windows 10

## GUI Version

v5.0.9

## Running standalone app

Running on Processing 4.0b3

## Console Log

[image]

retiutut, 2022-01-21:
@cfausn There is an open PR to bump the GUI to finally use Processing 4. I have linked this issue. Thanks for testing!

retiutut, 2022-01-21:
@cfausn The problem here is that Processing 4 beta 2 uses Java 11, while beta 3 goes all the way to Java 17 and has not been tested.

retiutut, 2022-01-21:
Replicated and Fixed!

closed 2022-02-02 by retiutut

retiutut, 2022-04-15:
I am now in the process of downgrading our work on 5.1.0 back to Processing 3. We had severe issues on Windows with the new Popup FilterUI Window when resizing and clicking inside this new window.

retiutut, 2022-04-15:
We will also need to explicitly say here on Github and the Docs that we need to use Processing 3.

Processing 4 is in beta as of today, but seems to have more issues than solutions for now.

PR #1019 (2022-02-01): Update to Processing 4 beta 3 CI pipeline
The purpose of this PR is to update the entire GUI to use Processing 4 and OpenJDK 11 (aka Java 11). Processing 4 is now in the beta stage and is much more stable than it was a year ago. I have been testing this periodically since the first Processing 4 release, and it seems it is finally time to make this change for the GUI.

- [x] Fixes #674 

- [x] Update BrainFlow jar to one that uses Java 11

- [x] Fixes #1025 

- [x] Test all BrainFlow boards in the GUI using updated BrainFlow jar
commit 2df2612d (2022-02-01): Processing4beta3 - Fix "type Serial is ambigious"
Fixes #1025