OpenBCI/OpenBCI_GUI #1026: Apply smoothing by default

ThreeForm, 2022-01-24:
## Problem

Inconsistent waveform update in GUI (known issue) when COM driver latency not changed from 16ms to 1ms and smoothing option is not on.

## Expected 

The software smoothing option should be on by default since that mostly fixes the problem. A "pro-tips" pop-up the first time the user starts the software can alert them to the fact they can shave a few hundred ms off the total latency (in addition to fixing the smoothness problem) by changing the latency setting.

## Operating System and Version

Win10

## GUI Version

5.09

## Running standalone app

downloaded binary

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

retiutut, 2022-02-02:
Since there is no way to confirm that the FTDI fix has been done via code, this would then be a hassle to turn off every time.

This is a "Session Setting" that applies only to cyton, which means it doesn't really belong as a "GUI-wide setting. Session settings are no longer loaded by default.

This is almost a lose-lose situation.

retiutut, 2022-02-02:
This does help Mac users and beginning users the most. I think this can be done if we increase the communication about the FTDI fix in the GUI, Docs, and maybe also BrainFlow documentation.

retiutut, 2022-03-08:
[image]

retiutut, 2022-03-11:
[image]

commit 24dc749c (2022-03-08): Show a popup to inform first-time Cyton users about the FTDI buffer fix and Cyton Smoothing feature. Fixes #1026
PR #1033 (2022-03-11): Set Cyton Data Smoothing to True by Default and Change Button Text
Fixes #1026 
Fixes #1027 

Update CI/CD to Processing 4 Beta 7 for all OS.
closed 2022-03-11 by retiutut