OpenBCI/OpenBCI_GUI #1029: Library files in home folder

cbrnr, 2022-02-07:
## Problem
The GUI puts the following files directly into the home directory (I tried this on macOS, but maybe this is also the case on Windows and Linux):

```
brainflow_svm.model
libBoardController.dylib
libDataHandler.dylib
libGanglionLib.dylib
libMLModule.dylib
libneurosdk-shared.dylib
```

## Expected 
I would expect that files required by OpenBCI are located in a more standard location. The program already creates a folder `~/Documents/OpenBCI_GUI`, so I was thinking that these files could be dumped into a subfolder in this location (e.g. `~/Documents/OpenBCI_GUI/Lib`).

## Operating System and Version

macOS 11.6.3

## GUI Version

v5.0.9

## Running standalone app

I'm running the standalone app.

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

retiutut, 2022-02-08:
@Andrey1994 Is this something that the BrainFlow library does by default for Mac OS X?

Andrey1994, 2022-02-08:
these files are extracted into the current directory or into a directory with an executable file and they must be in a search path, they cannot be in a folder like documents

cbrnr, 2022-02-08:
So can the current directory be set to e.g. the folder inside Documents?

retiutut, 2022-02-08:
Mac OS X prefers for all .app applications to be run from Applications folder.

cbrnr, 2022-02-08:
Yes, so usually app bundles include their libs. Would that be possible?

retiutut, 2022-02-08:
@Andrey1994 Can you shed light on why we need to unpack the libs to the Mac OS X Home folder? I can imagine this is done for a specific reason.

Can we close this issue? It doesn't really seem like a problem that needs to be fixed or changed.

cbrnr, 2022-02-09:
Please don't close it, I do think this is an issue. MacOS apps are not supposed to dump shared libraries into a user's home. Instead, they should bundle them inside their app folder (i.e. `OpenBCI_GUI.app`). AFAIK, these should go into `OpenBCI_GUI.app/Contents/Frameworks`. I don't know how this particular app bundle is created, but there has to be a way to tell the executables to look for these libraries in this subfolder.

cbrnr, 2022-02-09:
Not sure how dependencies are installed in Java, but it seems like there is a brainflow release which contains all dependencies: https://github.com/brainflow-dev/brainflow/releases

Maybe using that would solve the problem (it does contain the *.dylibs that seem to be required)?

retiutut, 2022-04-05:
[image]

@Andrey1994

retiutut, 2022-04-28:
We looked into this issue and determined this is done by BrainFlow. It does seem to work and the files are accessible by the GUI on Mac. As for the error I shared above, it was run under different circumstances, and we will be using Processing 3 for the near future until Processing 4 is more stable.

closed 2022-04-28 by retiutut