OpenBCI/OpenBCI_GUI #1030: Resizing GUI freezes the app

cbrnr, 2022-02-07:
## Problem
I cannot resize the window. Whenever I try, the program immediately freezes and I have to kill it.

## Expected 
I would like to resize the window, because the default size is way too small.

## Operating System and Version

macOS 11.6.3

## GUI Version

v5.0.9

## Running standalone app

I'm running the standalone app.

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

retiutut, 2022-02-08:
I believe this may be fixed by moving to Processing 4, which is planned for GUI 5.1.0 and already committed to the `development` branch.

cbrnr, 2022-02-09:
I tried, but I get an error related to libraries not found (the libraries that are normally in the home directory which I reported in #1029):

```
UnsatisfiedLinkError: Unable to load library 'libBoardController.dylib':
dlopen(libBoardController.dylib, 9): image not found
dlopen(libBoardController.dylib, 9): image not found
Native library (darwin/libBoardController.dylib) not found in resource path (/var/folders/6_/gnl5gd692xqbf4vd6pqvc_1c0000gn/T/OpenBCI_GUI12067625345381685595temp::/Users/clemens/Documents/Processing/OpenBCI_GUI/code/LSLLink.jar:/Applications/Processing.app/Contents/Java/core/library/jogl-all.jar:/Applications/Processing.app/Contents/Java/core/library/gluegen-rt.jar:/Applications/Processing.app/Contents/Java/core/library/core.jar:/Users/clemens/Documents/Processing/libraries/LSLLink/library/LSLLink.jar:/Users/clemens/Documents/Processing/libraries/LSLLink/library/jna.jar:/Users/clemens/Documents/Processing/libraries/LSLLink/library/jna-platform.jar:/Users/clemens/Documents/Processing/libraries/minim/library/tritonus_share.jar:/Users/clemens/Documents/Processing/libraries/minim/library/mp3spi1.9.5.jar:/Users/clemens/Documents/Processing/libraries/minim/library/jl1.0.1.jar:/Users/clemens/Documents/Processing/libraries/minim/library/minim.jar:/Users/clemens/Documents/Processing/libraries/minim/library/jsminim.jar:/Users/clemens/Documents/Processing/libraries/minim/library/tritonus_aos.jar:/Applications/Processing.app/Contents/Java/modes/java/libraries/serial/library/jssc.jar:/Applications/Processing.app/Contents/Java/modes/java/libraries/serial/library/serial.jar:/Applications/Processing.app/Contents/Java/modes/java/libraries/net/library/net.jar:/Users/clemens/Documents/Processing/libraries/grafica/library/grafica.jar:/Users/clemens/Documents/Processing/libraries/GifAnimation/library/GifAnimation.jar:/Users/clemens/Documents/Processing/libraries/oscP5/library/oscP5.jar:/Users/clemens/Documents/Processing/libraries/udp/library/udp.jar:/Users/clemens/Documents/Processing/libraries/jSerialComm/library/jSerialComm.jar:/Users/clemens/Documents/Processing/libraries/brainflow/library/brainflow.jar:/Users/clemens/Documents/Processing/libraries/httprequests_processing/library/commons-logging-1.1.1.jar:/Users/clemens/Documents/Processing/libraries/httprequests_processing/library/httpclient-4.1.2.jar:/Users/clemens/Documents/Processing/libraries/httprequests_processing/library/httpcore-4.1.2.jar:/Users/clemens/Documents/Processing/libraries/httprequests_processing/library/commons-codec-1.4.jar:/Users/clemens/Documents/Processing/libraries/httprequests_processing/library/httpclient-cache-4.1.2.jar:/Users/clemens/Documents/Processing/libraries/httprequests_processing/library/httprequests_processing.jar:/Users/clemens/Documents/Processing/libraries/httprequests_processing/library/httpmime-4.1.2.jar:/Users/clemens/Documents/Processing/libraries/controlP5/library/controlP5.jar:/Users/clemens/Documents/Processing/libraries/openbci_gui_helpers/library/openbci_gui_helpers.jar:/Users/clemens/Documents/Processing/libraries/ssdp_client/library/ssdp_client.jar)
```

retiutut, 2022-03-07:
@cbrnr Can you please share the full console log? This is not enough information to understand why resizing the app is related to a BrainFlow dll file.

retiutut, 2022-03-07:
I would try the latest Processing 4 build, run the GUI from this in the `development` branch, and share the full console log. https://github.com/processing/processing4/releases/tag/processing-1282-4.0b7

cbrnr, 2022-03-10:
I have an Intel Mac, so this is not related to an issue with Apple Silicon. As I've mentioned previously, starting the OpenBCI_GUI from the latest Processing build gives me an error because `libBoardController.dylib` cannot be found. Where can I get this library (and the other ones that are probably also missing)?

retiutut, 2022-04-19:
We will now be using Processing 3. This change applies to the current `master` branch and the `development` branch. Apple Silicon should be able to run the app as an Intel app. There were too many serious errors occurring in Processing 4 ahead of finally releasing 5.1.0.

retiutut, 2022-04-28:
Please try again using the `master`  branch and Processing 3.5.3. This should resolve your issue. If you still get an error, share the console log here and we can reopen the issue. Thank you for making this issue.