OpenBCI/OpenBCI_GUI #1042: OSC communication problem

thiago-roque07, 2022-04-20:
## Problem

I've been trying to send band power levels from OpenBCI GUI to Max/msp through OSC.
This was working fine until v5.05, but with both v5.08 and v5.09 I was not able to stablish stable communication.
All packets seem to be dropped.

By using python (python-osc), I was able to receive the packets from any version of OpenBCI, but for some reason, at max/msp this is not working.

## Expected 

Have a fully functional OSC communication on the most recent version of OpenBCI

## Operating System and Version

Windows 10 Student Edition

## GUI Version

v5.05, v5.02, v4.20 - fully working
v5.08 and 5.09 - not wokring

## Running standalone app

Standalone

## Type of OpenBCI Board

Indifferent

## Are you using a WiFi Shield?

Indifferent

retiutut, 2022-04-20:
Please share your MaxMSP program @thiago-roque07. This is best done in a simple Max sketch and sharing the compressed code here. We also need to know what version of Max you are using.

It's important to note that the format may have changed over time, and I'd like to see what objects and how you are receiving the data in MaxMSP.

thiago-roque07, 2022-04-20:
Hi Richard, 
this problem is happening even on the standard udpreceive example sketch.
I'm using Max v8.1.8
I've heard that Max have been having some security updates recently that is causing problems with network connections, so I've been a bit apprehensive about updating it.

retiutut, 2022-04-20:
I was not able to replicate this issue. Here is the Max Patch that I used:

```
<pre><code>
----------begin_max5_patcher----------
24901.3oc68t1jqibbnfedzuBrcbudsu6Ynq2UgMVGgzHe0r5CyZER5ZGNjc
zAZRzciY.InHAOmyLNt529VO.HAHdVfD.EYyVZ5Ca.Px7clUlYk0+0u5ad5k
juFt+Iu+u89Kdey27e8q9luQeI0E9lr+9adZcvWWFGrW+XOsI7KIu7iO8Iys
RC+Zp9xIK1kbHMz6eLYa3lWVFk+.aNrNZSbXp9MCytXzJy64ke7agLbgGU9Q
j+rnrqZtT5OuMz.kO8zm7d52cHN9ODr7mBke4+mYO31fzkuGs4sm2EtL07rb
fXA3Sd9fET4uopWiPK.d+mp2w+6e0uR8qOcY38+xe529sZLuVDFTCBSqGeAs
fFDlBzIbn5eXrKBOVGteevagUPD4C8WODdHbezuD5AApepEkP0fRn5QIXirv
1XZDglQIvpeCQ9iAWK88cG5qDp+0D4xjHgTMNhgiAxcX0V4WXXzmk7QDlv8V
tYcPZuUHACFe2m9yw5q9zpvWCNDe76rYBAQSGfDvXPH15869c+49h2H6UKMn
.NCSJfBEeOg6x.7LH+ad50n3vOGtaeTxlBvx27Tv1sEt72T3snP2eLQ+AI9z
wKEswbIzwKsK7yQ4uevwqFrSB4oRv9vNC24qLxSm9XRVEtaygHMnXtnjvmAR
ZR7l.oMisAKMuYEmH+1mnELpVTlh0pq9DikJewIpgjG9Vbhzj8phP2SJ2EQa
jBr6C2jFjdFvmKH87qIaRUVlzPojFW28eMCDq8lJjPC++lcQAwGwf21EsJYi
BHJwJTWN+q6uH0VM5rEQF8SrIXaMuYozjjtzvM2KQxC6eIXmhS8hQcImA9TZ
RRb4ac78EG9ZZ1s2FsYyYTwzjsMeycQu8dKu2WRj2bcae156r+4CaL28YoPQ
5y6C9bYpcZPbblhX4O9uFrIRZ.JLMxvBPfi2LbSfDQee+xcIwwkvWyc9bM2Y
kTHeY3WhVk9t9Kpnvf7wi1lKD8zQt7pn2B2mV9ZoAusu7UNY95zkN7RlR7yo
gq2FKwhxOfT8HZe592S9x9rGLWPqHA3TDVEUpKZUqz0Oy5l1r0QE1FBwnhY7
ydCUsj2g07lcOizF53Fae4+tzaaYx50R04RjJM7sYU3WKXnIyPelMmARdJY7
uQ27M5CnBgS.FHoKZSZeodDtXASFgEi4a9gwkjzyB0Y7oOeqGpQxCpAxCWLA
jGYXABI4AJ.lePXM4gNFjmCqeQ5i1VoDN6oRnxNoZepzLgwt04Jl0tzFSv.E
crIVT7MIghn0GVeNTHg9iWl0pprbgRuDr4sNo5TyRijTasWaSfL9SmXnYkiP
6EEIO0KZbsTldanCBqJKRXSsppLBIYnDq712hqfFjUQD9X4LfBI5v9jKYCxX
R4nbZj+mjqZcTnQ49VpiH8CAe06yAwGB8dM4fjXEj5gwKDjC+qK79tnMdFcc
OD06SdIp+cA1++2eo3GUbzlvkx2a54xP8iPSalPC5EkjA4KnbAvmjQI4BohI
y3rEOgjx+76gdaCC9oL5YfjZJQlsGR8hR++buWrLpGOs6bupFPKQDwVSDQCm
HxMDJFcACiQ7LZHvW53.B4FOuSmN6uDuPF.osl0PXwDXViHk2JQivRUVBZpM
qse6NYbAu5Uil6+8EnWOSw8+d7p+i+Cspq5lsn41nPGtIZ9nahT5PshIRnQd
k3Osxjwx0OYu6VoivwTtLiNgErEDFiwoGUdkDO9TKWpHSI6R891APlPSAYh5
u.KIND+bxDbzHSqODmFIEZV0yvjkgnHui7iv.w5vIgfyWiXw.4f9Wmfour.8
jjTVEMTe8hxX7yf9S4hhNQw+oxk4y+XTpjtjpRlgsKXwumKqs3BRPfRKHoB4
UBNOuNHcWzW6wZLzDRnIAgPBOO+2UeimxXZYZUiYNsormd1autrn1PlTaNap
cmQ0lxpZoLq1irqdVFV+VnPnW0A2PAETsTq+4hl0mm0djq09ju0Ny4ZOy6ZK
4dse4esibv1YdX6LWrcjO1tyIam4ksG4lsO4m0lbz1RdZ6LWssmu11yYa64s
s0b21T9aqOGtMjG2dkK2dlO25yo64lZN2hek62kk+xV+iSBVoJbqG37GoIa+
ma+mTyardO.8HN2lLxCgXsccn1cuzTUMqmnjqxxtKuThnAnGJEBA4sSi.8f.
3axgFjXJrY84Gusbj2PdxGQpVeE8fCmvhmBgOhIlBe3bH5IiDZw1uH4aIeYv
jIfe6jITWjoBgi8o9pwRI50S3SoKToB.hxWNgJW6RsXehufSxpRgORElLaAV
8Z7wKV4KZ+OuY4TI9dUn77YfxS.HisReSB2uEH1cXqPwKLjAOh2x2C1IUCAx
0WAFNmgNKblLuXTVg0rLY1RrlHCYp0g58qUk67BLSCXyoV.xDjOWLoz5WNHC
tdyvIYUBqpijWLDueUpgWyFywZi4bBdgp9An7hHnpdpoZLjIhtFjlt6PzPoq
P6hWnzhUwKNmbqfkr5LJWvW3W+4e4I6C3n80x1.CAALA.JiKFaH+FoabMwCp
zue93pizYKoJDT2xs0qDUtf+oxGvExZEhwg4lotM0bV.1X+hhajyVhwcCvf7
GGFz9sgKODGrapXQPNOiEk0JwTP+XQ2r5V3wgwsJ50WOrOb54aTwGC9FzmNN
btf0uDUt.+SDmCeynw0ifseKdQrJMsdp5t77qulhFNilcUYziFizGAKt1THD
dhQdCnNwGG0IMgdp0kflsW.xrNo6ZqfiTjgaS1Goqqvjy55Np+6CVGZjh7HP
UkkIOh97vNbmH5Kum8p5kZYbX03papYUpn2g4SRsgnHtpAt4bdkjSf42PI8A
hbpj9PXlDmgvh7L7Lg44oKIyvcA6CedYRbxNOvhr+ObwfkUQvooNlB8VFFhD
lls.MsBn8hpNTZHgNEjPJMixw7MsapqkB8sAK+IORqDQRKBhP6xUdIOdftCj
eHoRGap4LS5+RJ6ls1KfqVzhKrxPyQsJxHvbjnhGL2uFQHppNQWXMh3yGUWs
YhqP0m3pE0CptoRzmRUf2uVOPExj1890xEinZFV0+M3nsuPlfNzhdS9AltiE
hPl3g43IkjeoIy1RmcmxV8xfAUMg5lS.pUR366yoY0vmq1g7loQBDcSQM4Cj
ZdAKAuuDTnw.LkcSQPQyu3o.4u.cdq+.g.7bHfVcKoc1JckqyZWf2xfMdIeN
b2tnUgduE6onKQxk0UMcAsxJd4vK0zWumyjHH65KvNxtw.yoTVFYY.4pnQL7
wcNNVsywMMrBot9Lzfi6kDpaj5ORniUZ.i+4mU8k+zkGvLdFUfWvE9D4pIyZ
hKtJ1Ez8REiIiTEiS1k9dxzwsx55KpJ+eRULHLiaQI2UbKJZb3Vwga1+rLzx
oqILfL+7lu.s.wQDFLeqaAUrL1cCKajLI9ZvtmWFGsc5XXYadIJfs.AEEFpB
p8Zm.d2vvXiCCaS3jywLyd.nbwqKfL4ZwyiRDn7gItarJxE2I8bHjYJbkLNJ
xBHvG.7K1R+n6Fihb+6jxEyLIGWtNCUfGDBAUbG.jU5g6gFKDLRAdjj7SASW
SNwPhrYVY1RicGuV8qYlxVC8uVg5WgdZRbIkBA2uRg7V7yJrPscq+z0mSxHx
vGYXnHeW1.ONLrlrVd5JrMl3yQ14yLdwjjqJa4f6jczjxEt2udUz5iEQIXUv
1TOfpZJgx68svETNPHu9lkumr64uptU1q+Y0qSd808goxqK+.FL2kcApYnKu
hiCSEiXJCIyzbBDefKUulias0cgAq7d4kf33EqS97P4PWR7HnIq2N8QYqYNa
CWQHtVIz9qoJtPTnGFA7jFT790eNIVqRcHMYeZvNsp2gMQ+0CgUKkYSSvM6J
wF8FTYyGvxRfEOKUwlIEjKprsJ4WB2cIJaXvMg1lbAXhxpahGpaUaIoaRea4
[... 467 lines cut ...]
oms31LRZMYVKeIj0mdzgjZzAlVzAl+s5yfVyoBsgzfluL2O8qZdQ6kWvdQI3
FR04vYBmmVS6So4PImCJUl8lIbMo00klxg+o8xgzzjMpTR1Nuy9DPdwHpAz9
CG1sM1JfSpXRzm2RRFFfKLJdRcP50D3xOqaMuUy65zoq5Ifql7ownRgJMw63
XslkYhnfvUYzBlMsyO8qiO2wIj9yQaTVnCadRoWDykVCxjcOArgwgqqj8OgJ
NU8r2F.EByqHTf5vspQvUhQXFwLCxk+KipEQXBoNqe42VGy18hPrQdnUdOSf
.Xy2kTu07sxkvgpddM9spkRHlwktjKnsu.oLACO.AFXuDXxSt2CQlGhLMaiY
cvd0HgonIl0g62G7V3U2yZOYYm6dsexkis20tBw4xj5OrVEc4CB9.io7lxBW
3uDrZ4eqaKbhigzK3.pPXdkJlz1sv0OjzBCbYC7pNA3qL4szmMgAg.y5a7QL
RVkBgBUnW8xh70zGh8VygFpqOEXriWGY9sfnM+sqFUt5WtfAIDgoL39HXsRl
qBtojL+oqrnYGBlCINkgnTLv.FZmVopvsI2GMFlRe0GFnguhSOulAi9ZQn8u
qWOrYYt2jlCK6pKPN3H2yh4pTGubsSwwvxWwwnIfLJmXRLhz.L0XIgjIlxI2
WIaJie7VR7JmjYzV+xbGxMTmoY4mFX0yINln6yzv+zY+1ufI2KJgsmLdb1KZ
gqk6jrzqjNPQWoDL1OlleeWLScsEXdduvjbzNOWXEMYtW5JTa+sg3FjFRfbM
FAyxws7ZGe204GskuXYLgjlbj1SJtMtzpyqvIfhpQH8qNBdbw0Zw5m7fpxl5
emo+BuRIUUBm9clIBFWXpMAPPDvqZpXq13p0iRCsYUul.Hx0APrqCfDWG.ot
L.tK4KarVK4DbcxzDdTgPjyCgXmGBINLD9a+4.6ECAHepIANbNFixNCYgPH9
5Cf8X8TCXWCbUANjKCbXWF3HtLvQcSf6GjKpcSZf0pDBBfwLcUBCHANs5KyG
5Su9vX4cMS8fXG6TlAtKYttH.5VGAv25H.4VGAn25H.6VDAzaosuy9vFT4zP
uze7wlBhSUM120G59dGLnlJ6DvdCa8Z2+cMAPjqCfXWG.ItN.RcX.7OZuFxw
dQEdLUi.DVZc4ZBcaC1DF+uFkb774YTZ041ZdRvBzPZ1Y38VyNuyjk2uuy5O
YmLbcY2kQTGVpYqBmyMqGW5R0+ptdbUB1CWEkdEwHUYxuzZR2O7wr6DzacWE
Bseavxr1GN4qO8q9e+q9+G8JrerC
-----------end_max5_patcher-----------
</code></pre>
```

retiutut, 2022-04-20:
There is likely no issue with the GUI. As I have attempted to replicate using more than one version of the GUI, a MaxMSP sketch that has verified to work, checking the Networking Widget code changes over time, and Python-OSC.

thiago-roque07, 2022-04-20:
Ohh... I removed the OpenBCI GUI v50.9 from my computer. To avoid that problem.
I can try updating my MaxMsp to 8.2 and test it with OpenBCI v5.09

Which version of Max are you using?

retiutut, 2022-04-21:
@thiago-roque07 The Max version doesn't matter so much. I just needed to know you are using Max 8. Please try my above sketch and let me know the result.

mhrahmani, 2022-07-29:
@retiutut Hi, 
I think the root of all evil in most OSC issues inside OpenBCI is the "Multi Type" Message formatting and extra spaces. 

In OpenBCI, some "sequential" **OSC** data streams have an `INT` `FLOAT` format. 
So **Band Power** for example in a 3 electrode (3 channel) is an OSC message like:

`/address`+`/`+` `+`int(chan)`+` `+`float()`+` `+`float()`+ ... 

usually in console outputs looks like this
```
/openbci/ 1 8.0 7.8 3.4 0.0 9.8
/openbci/ 2 5.7 6.5 0.0 0.0 5.0
/openbci/ 3 4.5 6.7 0.0 0.0 0.0
```

These type of streams are the ones that make a lot of OSC parsers freak out and leak into memory and crash. I've seen it with **Accelerator/AUX** stream and **Band Power** stream. Take a look at the OSC packets:

[image]

The _mixed-type_ messages from Accelerator and Band Power will make OSC receivers _crash_. 
While _single-type_ messages from _Average Band Power_ and _Time Series_ will be _fine_. 

Note that **OSC Has Support for mixed-type streams** but for some reason **oscP5** library is doing it wrong. 

### Solution 1: 
Converting the `int` Channel Number value to `float` and passing it to oscP5 as `float` instead. 

```
/openbci/ 1.0 8.0 7.8 3.4 0.0 9.8
/openbci/ 2.0 5.7 6.5 0.0 0.0 5.0
/openbci/ 3.0 4.5 6.7 0.0 0.0 0.0
```

This is probably the least painful solution without having to dig into oscP5. 

### Solution 2: 
Appending the `Channel Number` to `Address` instead of passing it as an `Argument`. 

for the above example it would be:

```
/openbci/1/ 8.0 7.8 3.4 0.0 9.8
/openbci/2/ 5.7 6.5 0.0 0.0 5.0
/openbci/3/ 4.5 6.7 0.0 0.0 0.0
```

This is the preferred way for OSC protocol. 
OSC parsers _prefer_ channels of a _single type_, and all the categorization info as strings in the address. 

**Question** could you point me to where in the code I could modify how data is packaged? 
Like... the least likely place to break anything else in openBCI.

retiutut, 2022-08-10:
I like solution 2 👍

mhrahmani, 2022-08-11:
First of all, thank you very much for all the work you do to bring us these tools and platforms. 

I ended up digging into oscP5 library and official OSC specs to learn how exactly the data is packed and parsed, plus I learned some communication protocol lore along the way. It was a journey! 

I did come up with a working fix that should make most OSC parsers happy. here's how it works for `Accelerometer/AUX` and `Band Power`, and it should work for other stream setups in `w_networking.pde` as well. 

## Problem1: INT "surprise"

- Here's the part of code where packets are built and sent:
https://github.com/OpenBCI/OpenBCI_GUI/blob/52447b7c644bf94c73cf663fd0bb6a0f5b064a28/OpenBCI_GUI/W_Networking.pde#L1855-L1871

- The MAIN culprit is in `ln:1862` where the `int` for `channel number` is added to the `msg` object. 

https://github.com/OpenBCI/OpenBCI_GUI/blob/52447b7c644bf94c73cf663fd0bb6a0f5b064a28/OpenBCI_GUI/W_Networking.pde#L1861-L1862

- The `msg` is an `oscP5` object that is basically an OSC packet ready to be sent, which has variables for `address (address pattern)`, `data (arguments)`, `network(ip/port/etc)` 

- the oscP5 `clearArguments()` method in `msg.clearArguments();` clears the `data` part but keeps `address` and `network` 

https://github.com/OpenBCI/OpenBCI_GUI/blob/52447b7c644bf94c73cf663fd0bb6a0f5b064a28/OpenBCI_GUI/libraries/oscP5/src/oscP5/OscMessage.java#L175-L184

- There's another method for messages in oscP5 called `setAddrPattern()` that updates the `address` but keeps everything else:

https://github.com/OpenBCI/OpenBCI_GUI/blob/52447b7c644bf94c73cf663fd0bb6a0f5b064a28/OpenBCI_GUI/libraries/oscP5/src/oscP5/OscMessage.java#L231-L241

### Accelerometer Stream Hotfix
- Now check out how we can `setAddrPattern()` to add the `channel number` to the end of `address pattern` field.

```java
                for (int i = 0; i < NUM_ACCEL_DIMS; i++) {
                    msg.clearArguments();
                    // msg.add(i+1);
                    msg.setAddrPattern(address + "/" + (i+1) + "/");
```

- This accomplishes 2 things:
1. Removes `int` "superise" from normally `float` messages and makes OSC messages of a single type. Not all OSC parsers and packagers know how to pack/unpack a multi-type OSC message, thinking they hit a `newline` byte or something which causes memory leaks. In general just avoid multi-type messaging for OSC and put all the routing info inside address instead.

2. Makes better use of OSC's built-in `REST` style routing that's more standard in real-time systems. There IS support for other addressing styles in the protocol but I'm not sure if oscP5 can handles it. 

-This way the rest of the code stays the same.

```java
    void sendAccelerometerData() {
        // UNFILTERED & FILTERED, Accel data is not affected by filters anyways
        if (this.filter==false || this.filter==true) {
            // OSC
            if (this.protocol.equals("OSC")) {
                for (int i = 0; i < NUM_ACCEL_DIMS; i++) {
                    msg.clearArguments();
                    // msg.add(i+1);
                    msg.setAddrPattern(address + "/" + (i+1) + "/");

                    //ADD Accelerometer data
                    msg.add(w_accelerometer.getLastAccelVal(i));
                    // println(i + " | " + w_accelerometer.getLastAccelVal(i));
                    try {
                        this.osc.send(msg,this.netaddress);
                    } catch (Exception e) {
                        println(e.getMessage());
                        msg.setAddrPattern(address); // cleanup just in case this is the only instance.
                    }
                    msg.setAddrPattern(address); // cleanup just in case this is the only instance.
                }
```

- **NOTE** also added those two `msg.setAddrPattern(address);` just  to reset the `address pattern` back to the original state. (I'm not sure how scopes work in Java, I need a better Java IDE to see all that). Might be a good idea to add another `clearArguments()` at the end as well. 

## Problem2: interlaced AUX/Accelerator streams with the same address

[... 60 lines cut ...]

In Max/MSP (and other Signal-Flow environments) it is common practice to receive a list of different values and use `route` and `gate` objects to parse and send the signals to where they need. 

[image]

This is because Max was built on top of **MIDI** and the whole **Route~Gate** strategy was/is a way to make use of the 8-bit MIDI protocol to carry custom signals and commands with bangs and clocks, like they do with physical modular synth wires and patches. (or PureData or SuperCollider).

It's pretty much the only way to route things in low-level signal flow. You have only 8-bits per message and it has to contain both **Address** and **Data**.

**OSC** is made by joining **UDP** and **MIDI** to utilize mainly the **Addressing** part of the UDP (among other things). It prefers if you provide a whole verbose address space down to each sensor `float` reading. Single variable, verbose address.

In fact you're already using `REST` parsing when you pass a `string` to the address, so any extra routing is extra parsing.
Maybe we can use the available `UDP` addresses we already have.

There **IS** support for Remote Procedure Call `int` addressing style like we do in MIDI, to make the packets smaller. and even oscP5 mentions it here,

 if you pass an `int` to **replace** the address pattern using the same `msg.setAddrPattern()`

https://github.com/OpenBCI/OpenBCI_GUI/blob/52447b7c644bf94c73cf663fd0bb6a0f5b064a28/OpenBCI_GUI/libraries/oscP5/src/oscP5/OscMessage.java#L231-L241

But I couldn't get it to work yet. 

Anyway, the stream now runs gloriously smooth:

[image]

retiutut, 2023-06-13:
Trying to address this tomorrow.

retiutut, 2023-06-14:
Using https://opensoundcontrol.stanford.edu/files/osc-best-practices-final.pdf

and going with REST

retiutut, 2023-06-14:
Alright! @mhrahmani you are welcome to test the changes pushed today in https://github.com/OpenBCI/OpenBCI_GUI/tree/june2023-networking-refactor

Example OSC Output:

/openbci/accelerometer/x 0.51234

Though, I've really gone ahead and applied this type of REST-style api to every data type.

I found that having a trailing slash is incorrect and therefore made sure it's not there in the address. I've also removed the address field for OSC Networking in the widget. If people want to run two GUIs, they can send to different ips/port. We will maintain a rigid OSC address structure that I'll share our official documentation while testing for the next release.

mhrahmani, 2023-06-15:
@retiutut Perfect timing! I'm just about to work on an experiment and getting back in the groove.

I'll check it out and let you know.

retiutut, 2023-06-22:
Ok I've updated both FFT and TimeSeriesFilt and TImeSeriesRaw outputs to be /openbci/[datatypename]/ch0 0. 0. 0. 0. etc.

Vanilla Pure Data doesn't seem to be able to gracefully convert symbols/float/ints. I hope this `ch1` part of the address is ok... which clearly should be a string in other receivers

retiutut, 2023-06-22:
Are you using TouchOSC in the above gif/videos? @mhrahmani

closed 2023-06-26 by retiutut