OpenBCI/OpenBCI_GUI #1065: How to change the LED of D1 on the cyton board from blinking to lit (so that the cyton board can be recognized by OpenGUI).

jockbin, 2022-06-30:
## Problem

The cyton board is not recognized by OpenGUI. The LED on D1 is blinking, which is likely the cause of the problem. The error message when launching OpenGUI is "OpenBCI_GUI: This application is not being run with Administrator access. This could limit the ability to connect to devices or read/write files".

## Expected 

The LED on D1 should turn on and the EEG should be able to measure correctly.

## Operating System and Version

Windows11

## GUI Version

v5.1.0

## Running standalone app

Downloaded app 

## Type of OpenBCI Board

Cyton+Daisy

## Are you using a WiFi Shield?

No

## Console Log
OpenBCI_GUI: This application is not being run with Administrator access. This could limit the ability to connect to devices or read/write files

retiutut, 2022-08-10:
Hello, there is not enough information here to describe the problem you may be facing.  Issues here are related to bugs or feature requests for the GUI.

```
[ERROR]: OpenBCI_GUI: This application is not being run with Administrator access. This could limit the ability to connect to devices or read/write files.
```

You should still be able to connect to the Cyton board on Windows 10 even though this message has appeared. You can also try running the GUI with administrator access.

For help with connecting to the board, please visit https://OpenBCI.com/forum.

retiutut, 2022-08-10:
I notice that the Console Log says Windows 10, yet you say you are using Windows 11.

closed 2022-08-15 by retiutut