OpenBCI/OpenBCI_GUI #1078: Cyton: EEG amplitudes are different in GUI and exported bdf file (scaling issues?)

LeXing1105, 2022-08-22:
## Problem

When I used the Cyton board to record an EEG/ECG signal, the signal amplitude in the OpenBCI GUI looks correct in the normal range, but when I exported the raw data into BDF or TXT file formats, the amplitude becomes very low (lower than in GUI), which is not correct. What caused these issues? Are there any scaling issues?  (wrong gain values? or similar)

In the GUI, I know there is a notch filter and a bandpass filter applied, In exported data, no processing applied, but the signal amplitudes differ too much!

The examples attached show the problem. 
EEG: in GUI and MatLab are the same recording,
ECG: in GUI and MATLAB are the same recording;

[image]
[image]
[image]
[image]

## Expected 

I want to know what caused this issue if there is something you already know, please help me foe debugging, if this is a new bug (amplification/scaling issue in your export code), please fix it and let me know what is the scaling factor, how to recover the amplitude back to the correct one.

## Operating System and Version

Windows

## GUI Version

GUI Version v5.1.0

## Running standalone app

check data in MATLAB EEGlab toolbox

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

## Console Log

Paste any relevant text from the console window here

retiutut, 2022-08-22:
Everything looks fine. The values in the GUI are in uVrms (microvolts root mean square).

retiutut, 2022-08-22:
@LeXing1105 Did you change the gain in the Hardware Settings?

closed 2022-08-22 by retiutut

LeXing1105, 2022-08-23:
@retiutut Thank you for your reply! The GUI figures may not be very clear in the post. For example, in the ECG signal, the 2nd channel has a ±400μV window, the ECG=93.3μVrms, but the peak value should be at around 400μV, right？ but in the Matlab plot, the exported ECG signal (from the same recording) only has a 10μV (between the baseline to the peak value), maybe the unit here is not μV, so it looks wired. The gain in GUI hardware settings is set as x24, as default.

The EEG signal and ECG have the same issue. What do you think the problem is?

retiutut, 2022-08-23:
It’s not very fair to say the values in the GUI are wrong without mentioning the units in MATLAB. Thanks for opening this issue but I still conclude there is nothing wrong.

LeXing1105, 2022-08-23:
@retiutut OK, fine, the MATLAB figures just plotted the raw exported data, without any processing. I am also not sure what the unit is. Can you please tell me what the unit should be from the raw exported BDF data?

retiutut, 2022-08-23:
The values stored to CSV and BDF are microvolts.

retiutut, 2022-08-23:
From DataLogger.pde

```
    private void saveNewData() {
        //If data is available, save to playback file...
        if(!settings.isLogFileOpen()) {
            return;
        }

        double[][] newData = currentBoard.getFrameData();

        switch (outputDataSource) {
            case OUTPUT_SOURCE_ODF:
                fileWriterODF.append(newData);
                break;
            case OUTPUT_SOURCE_BDF:
                fileWriterBDF.writeRawData_dataPacket(newData);
                break;
            case OUTPUT_SOURCE_NONE:
            default:
                // Do nothing...
                break;
        }
    }
    ```
    
    Here we can see we simply get data between the last frame and the current frame, and this is passed either to CSV or BDF writer class, and is unfiltered.