OpenBCI/OpenBCI_GUI #1079: MacOS X FTDI Latency Fix

retiutut, 2022-08-31:
## Problem

The default latency for the FTDI driver for Max OS X is 16 ms. This makes data appear choppy both in BrainFlow and the GUI.

## Expected 

It needs to be 1 ms to make Cyton and CytonDaisy data appear smooth for data streamed through BrainFlow and the GUI.

## Operating System and Version

Mac OS X 10+

## GUI Version

GUI 5.1.0

## Running standalone app

Standalone and from Processing

## Type of OpenBCI Board

Cyton and Cyton+Daisy

## Are you using a WiFi Shield?

No

## Console Log

N/A

https://openbci.com/forum/index.php?p=/discussion/comment/18232#Comment_18232

retiutut, 2022-08-31:
@Andrey1994 Maybe this fix can be embedded in BrainFlow specifically for Cyton, CytonDaisy and Mac.

Andrey1994, 2022-09-15:
need help with testing https://github.com/brainflow-dev/brainflow/pull/551/files

wjcroft, 2022-10-19:
Suggest adding a comment on the ioctl in your source, to state that the argument is 1 millisecond, not 1 microsecond. This may be an idiosyncrasy of the FTDI chip implementation. And Apple's ioctl man page or documentation is probably correct in stating that the original intention is microseconds.

See Scott Hannahs' comment here, shout out to Scott,

https://openbci.com/forum/index.php?p=/discussion/comment/18231/#Comment_18231

https://nationalmaglab.org/about/organization/scott-hannahs-lab-leadership

retiutut, 2022-12-13:
https://github.com/OpenBCI/Documentation/pull/134

closed 2022-12-13 by retiutut