OpenBCI/OpenBCI_GUI #1080: macOS GUI issue

TenTennyson, 2022-09-02:
## Problem

Slow data showing in GUI v5.1.0 on macOS 

## Expected 

Data will resume normal expectancy that is showed in Windows

## Operating System and Version

macOS Big Sur
Version 11.6
MacBook Pro (13-inch, M1, 2020)
Chip Apple M1
Memory 8GB

## GUI Version

GUI v5.1.0

## Running standalone app

Running the downloaded app

## Type of OpenBCI Board
Cyton and Dongle

## Are you using a WiFi Shield?

No

## Console Log

retiutut, 2022-09-22:
[image]

retiutut, 2022-09-22:
Does not work on this Mac
[image]

retiutut, 2022-09-22:
Does work on this Mac
[image]

retiutut, 2022-10-20:
https://github.com/brainflow-dev/brainflow/issues/568

retiutut, 2022-10-20:
This issue occurs when plugging in the BLED112 Dongle (USB Type A) into a USB C Hub or a USB Type C to Type A converter. We see a slower sampling rate which ranges from 82Hz (worst case) to 195Hz (best case). The desired sampling rate is 200Hz. An issue has been logged at the BrainFlow level to see if we can go back to using the built-in bluetooth on Macs. This was possible with the now-deprecate OpenBCI Hub (GUI v4 and NodeJS). But now, Apple silicon exists and we have the BrainFlow backend. This issue has been fully investigated with quite a few USB C adapters and hubs and we believe this is the path forward to resolve this for Mac users.

wjcroft, 2022-10-23:
**Another perspective on the assumption that this is caused by ‘bad’ USB-C to USB-A adapters.**

Instead, the slight USB system timing differences induced by these adapters, may be aggravating a BLED112 protocol / timing bug. And could be solved by refactoring how the BLED serial port interfaces is being interacted with. As you might know, I believe AJ Keller wrote the first version of this BLED code for the original ‘Hub’ implementation. I believe that code was then adapted by Andrey to offer BLED interface in Brainflow. But likely with minimal BLED serial protocol mods.
The BLED serial port protocol is quite complex and likely AJ just lucked out in the original implementation that more timing variances were not present in the earlier generations of USB.
Some monitoring and analysis of the USB serial port commands / responses with the BLED, may turn up the source of these glitches.

wjcroft, 2022-10-24:
Just as an offhand guesstimate, if the original Mac AJ serial interface code never expected to receive more than one input packet per cycle -- a backlog could be created if an excess packet was available but not picked up. With the slight timing changes induced by the USB-C adapter, such 'missed' input packets would therefore backup inside the BLED112.

wjcroft, 2022-10-24:
I have not looked at AJ’s BLED packet receive code, but the situation may be as simple as this: Inside the BLED112, available packets are added to the tail of a delivery queue as they are received. When the Mac code polls for an available packet, it may only ask for one and assume that is all that will ever be available. If timing is such that more than one packet is available, but not asked for — then they will pile up inside the BLED112 delivery queue.

Andrey1994, 2022-10-31:
I believe it will solve the issue partially https://github.com/brainflow-dev/brainflow/pull/574

wjcroft, 2022-11-01:
Matt, as I mentioned on the Forum thread, you cannot infer your sample rate from the 190 Hz rate the USB-C port is providing you. Your sample rate is very close to 200 Hz, regardless of how much delay is imposed by the usb pipeline. The sample rate stays constant at 200.x Hz, as the 'delay' to receive the samples creeps up continuously (10 samples per second). 

https://openbci.com/forum/index.php?p=/discussion/comment/18435/#Comment_18435

Yes, it certainly looks like the buffer in the BLED112 is accumulating these delayed samples. @Andrey1994 informed me that each sample should be accompanied by a BLE 'notification'. But if these notifications happen to be occasionally lost, then the sample will not be collected immediately, but instead will pile up in the buffer, to be eventually collected after some growing delay factor.

wjcroft, 2022-11-01:
If you are delivering the stimulation via a monitor flashing pattern, there has been much written about how to be sure this is calibrated correctly. In particular the links I gave to the MindAffect calibration tools.

retiutut, 2022-12-06:
Hey everyone! We have a great solution coming down the pipe with native BLE access for Ganglion and Mac!

retiutut, 2023-02-02:
The fix for this issue has been merged into the `development` branch! The next release of the GUI will support connecting to Ganglion using built-in bluetooth! Thank you everyone for your patience and assistance with this issue. Shoutout to @Andrey1994 for implementing SimpleBLE on the BrainFlow side.

closed 2023-02-02 by retiutut