OpenBCI/OpenBCI_GUI #1081: EEG signal amplitude looks not correct in the exported files

LeXing1105, 2022-09-05:
## Problem

When I used the Cyton Board to record an EEG signal, the EEG signal in the OPENBCI GUI has an amplitude of around 10-30 uVrms (at this level). I set up the hardware gain as default x24. and applied bandpass filter (1-50 Hz, 4-order Butterworth filter), and a Notch Filter (50 Hz). The real-time EEG signal looks very good to me, having a reasonable amplitude level after the preprocessing/filtering.

However, when I opened the exported BDF file of these EEG recordings, I did exactly the same filtering (1-50 bandpass filtering- 4-order Butterworth, and 50 Hz notch filtering) in MATLAB script. The resulting filtered EEG signal has a very low amplitude at below 1 uVrms. (This is not correct for normal EEG amplitude range, and this is not identical to the OpenBCI GUI amplitude level, about 10 times lower than GUI)

I have attached 3 pictures:
(1). The example signal in the GUI (Screenshot)---10.6 uVrms
(2). The raw OpenBCI data (1-minute EEG recording, plotted directly from the raw exported BDF, it has powerline noise and baseline wandering, which is very reasonable, it has 28.1779 uVrms before filtering)
(3). The filtered EEG data (exactly the same filters as used in the GUI, the baseline wandering and 50 Hz powerline noise were removed, and the signal quality is good, but the amplitude is super low, not at the similar amplitude level as in the GUI, only 0.4120 uVrms now)

**Do you think are there any bugs in "data export code" in the GUI?** What do you think the problem is? What do you suggest to solve the issues?

Please help!

## Expected 

I want the exported EEG signals (after the same filtering as in the GUI) to have a similar amplitude level as in the GUI, to have a reasonable EEG amplitude.

## Operating System and Version

Windows

## GUI Version

GUI Version v5.1.0

## Running standalone app

## Type of OpenBCI Board

Cyton

## Are you using a WiFi Shield?

No

## Console Log

Paste any relevant text from the console window here
[image]
[image]
[image]

retiutut, 2022-09-06:
https://github.com/OpenBCI/OpenBCI_GUI/issues/1078

LeXing1105, 2022-09-07:
@retiutut Hi, I think I didn't explain clearly last time in #1078. 

In this post here, I made it clearer, Don't you think there are some issues? Can you explain why the exported signal has such a low amplitude? (much smaller than in GUI). They are actually from the same EEG recording! I still don't understand why? If you know, please clarify it! Thank you very much!

retiutut, 2022-10-20:
https://github.com/OpenBCI/OpenBCI_GUI/issues/1078#issuecomment-1224974081

closed 2022-10-20 by retiutut