OpenBCI/OpenBCI_GUI #1086: GUI will not launch on Ubuntu 20.04.4 LTS

philippitts, 2022-11-03:
## Problem

GUI will not launch on Ubuntu 20.04.4 LTS.

## Expected 

The GUI will launch and run as normal.

## Operating System and Version

Ubuntu 20.04.4 LTS

## GUI Version

5.0.8

## Running standalone app

Downloaded app

## Type of OpenBCI Board

N/A

## Are you using a WiFi Shield?

N/A

## Console Log

```
sudo ./OpenBCI_GUI 
libEGL warning: DRI2: failed to authenticate
OpenGL error 1280 at bot beginDraw(): invalid enumerant
Console Log Started at Local Time: 2022-11-03_17-17-26
Screen Resolution: 2560 X 1600
High-DPI Screen Detected: false
Operating System and Version: Linux - 5.13.0-30-generic
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://docs.openbci.com/Software/OpenBCISoftware/GUIDocs/
OpenBCI_GUI::Setup: GUI v5 Sample Data exists in Documents folder.
Settings: LogFileMaxDuration = 60 minutes
OpenBCI_GUI::Settings: Found and loaded existing GUI-wide Settings from file.
ControlP5 2.3.4 infos, comments, questions at https://github.com/retiutut/controlp5
OpenGL error 1280 at top endDraw(): invalid enumerant
TopNav: Internet Connection Successful
Local Version: v5.1.0, Latest Version: v5.1.0
GUI is up to date!
BrainFlow Version: fce5573c9456f5216a457fc66171b28a175e8b95
OpenBCI_GUI: AuditoryFeedback: Loading Audio...
==== JavaSound Minim Error ====
==== Couldn't open the line: line with format PCM_SIGNED 44100.0 Hz, 16 bit, stereo, 4 bytes/frame, little-endian not supported.

java.lang.NullPointerException
	at ddf.minim.ugens.FilePlayer.<init>(FilePlayer.java:48)
	at OpenBCI_GUI.asyncLoadAudioFiles(OpenBCI_GUI.java:2175)
	at OpenBCI_GUI.delayedSetup(OpenBCI_GUI.java:622)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at processing.core.PApplet.method(PApplet.java:3860)
	at processing.core.PApplet$2.run(PApplet.java:3898)
SHUTDOWN HOOK
X11Util.Display: Shutdown (JVM shutdown: true, open (no close attempt): 2/2, reusable (open, marked uncloseable): 0, pending (open in creation order): 2)
X11Util: Open X11 Display Connections: 2
X11Util: Open[0]: NamedX11Display[:0, 0x7f455c0012b0, refCount 1, unCloseable false]
X11Util: Open[1]: NamedX11Display[:0, 0x7f455c0159a0, refCount 1, unCloseable false]
```

retiutut, 2023-01-18:
Unable to replicate on Ubuntu 20.04.5 LTS with Radeon graphics card.

```
```
Console Log Started at Local Time: 2023-01-18_15-37-33
Screen Resolution: 2560 X 1440
High-DPI Screen Detected: false
Operating System and Version: Linux - 5.15.0-58-generic
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://docs.openbci.com/Software/OpenBCISoftware/GUIDocs/
OpenBCI_GUI::Setup: GUI v5 Sample Data exists in Documents folder.
Settings: LogFileMaxDuration = 60 minutes
OpenBCI_GUI::Settings: Creating new GUI-wide Settings file.
OpenBCI_GUI::Settings: Successfully saved GUI-wide settings to file!
ControlP5 2.3.4 infos, comments, questions at https://github.com/retiutut/controlp5
TopNav: Internet Connection Successful
Local Version: v5.1.0, Latest Version: v5.1.0
GUI is up to date!
BrainFlow Version: fce5573c9456f5216a457fc66171b28a175e8b95
OpenBCI_GUI: AuditoryFeedback: Loading Audio...
OpenBCI_GUI: AuditoryFeedback: Done Loading Audio!
CP: Playback History Size = 3
OpenBCI_GUI::Setup: Setup is complete!
[ERROR]: OpenBCI_GUI: This application is not being run with Administrator access. This could limit the ability to connect to devices or read/write files.
sourceList: click! 
OpenBCI_GUI: Channel count set to 8
[DEFAULT]: Attempting to Start Session...
initButtonPressed: Calling initSystem() after next draw()
OpenBCI_GUI: Start session. Calling initSystem().

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

OpenBCI_GUI: Init session using Synthetic data source
OpenBCI_GUI: Channel count set to 8
"Arial" is not available, so another font will be used. Use PFont.list() to show available fonts.
InitSettings: Saving Default Settings to file!
SessionSettings: 3 active widgets saved!
[SUCCESS]: Session started!
New widget [3] selected for container...
Your cursor is in FFT Plot
Need to call beginShape() first
[DEFAULT]: openBCI_GUI: startButton was pressed. Starting data transfer, wait a few seconds.
OpenBCI_GUI: openNewLogFile: opened ODF output file: /home/retiutut/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2023-01-18_15-37-41/OpenBCI-RAW-2023-01-18_15-37-56.txt
OpenBCI_GUI: BrainFlow Streamer Location: file:///home/retiutut/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2023-01-18_15-37-41//BrainFlow-RAW_2023-01-18_15-37-41_0.csv:w
Brainflow start streaming
[DEFAULT]: Data stream started.
[5] onOff released - Off
[6] onOff released - Off
[7] onOff released - Off
[8] onOff released - Off
ConsoleLog: Copying console log to clipboard!

```

philippitts, 2023-01-18:
When I encountered the issue, I was running off a live boot USB drive. That may be the problem here so I'll perform a full installation, retry, and report back.

philippitts, 2023-02-03:
This may be resolved by https://github.com/OpenBCI/OpenBCI_GUI/pull/1115, but needs to be tested.

philippitts, 2023-02-21:
> This may be resolved by #1115, but needs to be tested.

#1115 appears to be fix the issue on Ubuntu 20.04. @poppingtonic can you confirm for your distro? You can checkout the [development](https://github.com/OpenBCI/OpenBCI_GUI/tree/development) branch and [run the GUI from processing](https://docs.openbci.com/Software/OpenBCISoftware/GUIDocs/#running-the-openbci-gui-from-the-processing-ide).

philippitts, 2023-02-27:
This issue appears to be resolved by #1115. Closing.

closed 2023-02-27 by philippitts