OpenBCI/OpenBCI_GUI #1093: Start data stream at the start of session

scott-OpenBCI, 2022-12-19:
## **Is your feature request related to a problem? Please describe.**
Possible easier/faster UI/UX. 

## **Describe the solution you'd like**
Instead of pressing the start stream button or pressing the space bar, start the data stream immediately. 

## **Describe alternatives you've considered**
None

## **Additional context**
Add any other context or screenshots about the feature request here.

philippitts, 2022-12-19:
This issue will affect the core workflow of the GUI, so it's worth discussing if we want to implement this issue before schedule it.

Pros:
- Fewer button clicks

Cons:
- Users may already be used to the existing way of doing things and changing it could be disruptive
- Having separate actions for starting the session and starting streaming makes it easier to understand where issue is if there's a problem with connection
- Automatically starting streaming may create extra recordings that will have to be discarded if the experiment requires specific hardware settings
- Application may appear to hang longer as it performs both the start session and start streaming actions at the same time.

retiutut, 2023-01-06:
Could be a setting that's toggled in the Control Panel, which can be persisted as a "GUI-wide" setting in the GUI settings class/saved file.

retiutut, 2025-03-31:
Fixed in https://github.com/OpenBCI/OpenBCI_GUI/pull/1239

PR #1239 (2025-04-01): GUI v7.0.0 - Alpha 0
Fixes many things

___

Known fixes:

- Fixes #1205 
- Fixes #1093 
- Fixes #1226 
- Fixes #1198
closed 2025-03-31 by retiutut