OpenBCI/OpenBCI_GUI #1094: Target widget must be open in order to stream from networking widget

philippitts, 2023-01-05:
## Problem

Data will not stream from the networking widget unless the widget that is the source of that data is open. For example, it is not possible to stream average band powers unless the band power widget is open.

## Expected 

Data will stream from the networking widget regardless if the widget that is the source of that data is open.

retiutut, 2023-01-06:
Drawing is the most expensive thing we do in Processing GUI. It shouldn't be that big of a hit to performance to compute all metrics in the background.

retiutut, 2023-08-28:
Networking Outputs, and those that need data processed in the background:

- Time Series filtered - Done
- Time Series unfiltered - Done
- **Focus - NEED**
- FFT - Done
- EMG - Done
**- AvgBandPower - NEED**
**- BandPower - NEED**
**- Accel/Aux - NEED**
**- Pulse - NEED**
**- EMG Joystick - NEED**
- Marker - Done

philippitts, 2023-09-07:
Closed by https://github.com/OpenBCI/OpenBCI_GUI/pull/1181

closed 2023-09-11 by retiutut