OpenBCI/OpenBCI_GUI #1097: Filter out .tty serial ports in Networking Widget Serial list

retiutut, 2023-01-06:
## Problem

Sometimes the list is too long, and Arduino IDE does this by default. This will make it much easier for new users to control an Arduino using the GUI. **Users may still need to scroll, but this will reduce the number of items by half for all Mac users.**

## Expected 

Shorten the list to only include .cu instead of .tty addresses.

## Operating System and Version

Mac

## GUI Version

5.1.1-alpha.0

## Running standalone app

Both

## Type of OpenBCI Board

N/A

## Are you using a WiFi Shield?

N/A

## Console Log

[image]

kkashiva, 2023-05-16:
Hi @retiutut I'd like to contribute to this as my first issue. 
My approach is to filter the **comPorts** ArrayList to only include .cu instead of .tty addresses, so that when the method **createPortDropdown** is called, only the filtered comPorts list is passed as argument.

Slight confusion on what the role of **cp5_networking_portName** is here, if you can guide me please?

retiutut, 2023-05-16:
https://github.com/OpenBCI/OpenBCI_GUI/blob/d9f9312e39364a019d0cfd5390eb4c774218be43/OpenBCI_GUI/ControlPanel.pde#L653

I would use the third party serial port library that's used in this method. We do want to see other ports besides the Cyton dongle though. But this method could be reworked to provide the solution with the output as a LinkedList.

cp5_networking_portname is a Cp5 instance that holds the dropdown object for the port names.

To complete this issue, simply make the new method and replace the list of com ports found here: https://github.com/OpenBCI/OpenBCI_GUI/blob/d9f9312e39364a019d0cfd5390eb4c774218be43/OpenBCI_GUI/W_Networking.pde#L157

Hope this helps,
RW

kkashiva, 2023-05-21:
Thank you for the guidance @retiutut that was helpful. Please review the pull request created here https://github.com/OpenBCI/OpenBCI_GUI/pull/1136#issue-1718585094

closed 2023-06-08 by retiutut