OpenBCI/OpenBCI_GUI #1098: Networking LSL AvgBandPower: only one band value is sent

chris-zielinski, 2023-01-11:
## Problem

When using the LSL networking protocol to send the AvgBandPower, the data output is only one single value.

## Expected 

Like for the other protocol, the expected output should be the 5 avg power value (one per band).
In the W_Networking.pde/sendNormalizedPowerBandData function,  the loop across each frequency band is missing for the LSL case.

## Operating System and Version
Windows

## GUI Version
5.1.0

## Running standalone app
downloaded app

## Type of OpenBCI Board
Synthetic

## Are you using a WiFi Shield?
No

## Console Log

Thank you!

retiutut, 2023-01-12:
Confirmed using Networking test kit. When this metric was added it was in support of OpenBCI team using UDP. Thanks for bringing this oversight to our attention, we will include the fix in the next version of the GUI.

retiutut, 2023-01-12:
Actually the code in `sendNormalizedPowerBandData()` is fine, since we pass in a single dimension array, which is a valid option for LSL push_sample. What's wrong is the number of channels per sample, which should be 5 instead of 1.

[image]

PR #1099 (2023-01-18): Networking LSL avgbandpower only one band value is sent
Fixes #1098 

[image]
closed 2023-01-18 by retiutut