OpenBCI/OpenBCI_GUI #1104: Brainflow Streaming does not interpret file:/// URI and Outputs to Directory

stellarpower, 2023-01-15:
## Problem

I presume from [`generateBrainFlowStreamerFileName()`](https://github.com/OpenBCI/OpenBCI_GUI/blob/master/OpenBCI_GUI/DataWriterBF.pde#L71) that the streamer is meant to create a File URI that is then interpreted somewhere along the chain to output at the correct location - be this in Brainflow or in the GUI code/helper libs.

Instead I get a directory written into the installation root of the GUI called `file:`, and the other elements of the URI have been interpreted literally - the `:w` specifier is included in the directory names; the slashes (including the leading ones as part of the protocol) interpreted as ordinary directory separators. I thus also get a nested tree of where the GUI originally wanted to output my data:

[image]

This isn't some sort of symlink or bind mount I've set up either; those are ordinary directory entries.

## Expected 

I'm not sure what default are expected, as I haven't played with the Brainflow streamer. But presumably the intention is to output to `OpenBCI_GUI` under my XDG `Documents` folder as with the CSV/BDF+ recordings.

## Operating System and Version

Ubuntu 20.04 AMD64

## GUI Version

Release 5.1.0

## Running standalone app

Standalone

## Type of OpenBCI Board

Reproducible with Synthetic

## Console Log

```
Console Log Started at Local Time: 2023-01-15_21-20-54
Screen Resolution: 1600 X 900
High-DPI Screen Detected: false
Operating System and Version: Linux - 5.15.0-46-generic
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://docs.openbci.com/Software/OpenBCISoftware/GUIDocs/
OpenBCI_GUI::Setup: GUI v5 Sample Data exists in Documents folder.
Settings: LogFileMaxDuration = 60 minutes
OpenBCI_GUI::Settings: Found and loaded existing GUI-wide Settings from file.
ControlP5 2.3.4 infos, comments, questions at https://github.com/retiutut/controlp5
TopNav: Internet Connection Successful
Local Version: v5.1.0, Latest Version: v5.1.0
GUI is up to date!
BrainFlow Version: fce5573c9456f5216a457fc66171b28a175e8b95
OpenBCI_GUI: AuditoryFeedback: Loading Audio...
OpenBCI_GUI: AuditoryFeedback: Done Loading Audio!
OpenBCI_GUI::Control Panel: Playback history file not found.
OpenBCI_GUI::Setup: Setup is complete!
[ERROR]: OpenBCI_GUI: This application is not being run with Administrator access. This could limit the ability to connect to devices or read/write files.
sourceList: click! 
OpenBCI_GUI: Channel count set to 8
OpenBCI_GUI: Channel count set to 4
[DEFAULT]: Attempting to Start Session...
initButtonPressed: Calling initSystem() after next draw()
OpenBCI_GUI: Start session. Calling initSystem().

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

OpenBCI_GUI: Init session using Synthetic data source
OpenBCI_GUI: Channel count set to 4
"Arial" is not available, so another font will be used. Use PFont.list() to show available fonts.
InitSettings: Saving Default Settings to file!
SessionSettings: 3 active widgets saved!
[SUCCESS]: Session started!
[DEFAULT]: openBCI_GUI: startButton was pressed. Starting data transfer, wait a few seconds.
OpenBCI_GUI: openNewLogFile: opened ODF output file: /home/ben/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2023-01-15_21-21-09/OpenBCI-RAW-2023-01-15_21-21-15.txt
OpenBCI_GUI: BrainFlow Streamer Location: file:///home/ben/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2023-01-15_21-21-09//BrainFlow-RAW_2023-01-15_21-21-09_0.csv:w
Brainflow start streaming
[DEFAULT]: Data stream started.
[DEFAULT]: openBCI_GUI: stopButton was pressed. Stopping data transfer, wait a few seconds.
Brainflow stop streaming
[DEFAULT]: Data stream stopped.
ConsoleLog: Copying console log to clipboard!
ConsoleLog: Copying console log to clipboard!

```

stellarpower, 2023-01-16:
Update: Got a bit lost previously, but following a tidy-up of old files, this is further complicated by the fact that the `file:` directory only contains subfolders. The actual files do end up under my `Documents` folder; it's just the directory structure that is created.

retiutut, 2023-01-17:
```
OpenBCI_GUI: openNewLogFile: opened ODF output file: /home/ben/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2023-01-15_21-21-09/OpenBCI-RAW-2023-01-15_21-21-15.txt
OpenBCI_GUI: BrainFlow Streamer Location: file:///home/ben/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2023-01-15_21-21-09//BrainFlow-RAW_2023-01-15_21-21-09_0.csv:w
```
These two files should end up next to each other. Is this the case @stellarpower?

Thanks for sharing this Console Log so that we can see the behavior of this feature on Ubuntu 20.

Rexblk81, 2023-01-17:
Know anything about gangstalker?.

stellarpower, 2023-01-19:
Apologies for the delay; I'd symlinked things to tidy it up a bit on my main machine, so had to confirm on my laptop.

I do get two files next to each other:
```
user@host ~> ls -lah  ~/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2023-01-19_01-49-41/
total 107M
drwxrwxr-x  2 user user 4.0K Jan 19 01:51 .
drwxrwxr-x 21 user user 4.0K Jan 19 01:51 ..
-rw-rw-r--  1 user user  49M Jan 19 02:05 BrainFlow-RAW_2023-01-19_01-49-41_0.csv
-rw-rw-r--  1 user user  58M Jan 19 02:00 OpenBCI-RAW-2023-01-19_01-51-57.txt
```
and then also
```
user@host ~> file  $InstallationRoot/file:/home/arcadia/Documents/OpenBCI_GUI/Recordings/OpenBCISession_2023-01-19_01-49-41/BrainFlow-RAW_2023-01-19_01-49-41_0.csv:w/  
....: directory
```
as an empty directory.

It looks like Brainflow is (I presume correctly) interpreting the File URI and spitting out the file there. But something else along the way has decided that it's a normal path and tried to make the parent directories. I'm only glossing over, but I think this may be your culprit here:
https://github.com/OpenBCI/OpenBCI_GUI/blob/master/OpenBCI_GUI/DataWriterBF.pde#L65

`fileName` is being set to a URI, but then we go ahead and make it on the filesystem like a regular file. I'd probably suggest keeping the filename as one subobject, and then caching the URI form, which would only be used (or regenerated) when needed from the canonical form, being a local path.

I'm interested if this doesn't cause issues on other platforms though. Just from what I can see in my browser I'd have expected that to behave the same across most OSs. Unless they are interpreting the URI into a path first - wouldn't be too surprised if this happened on Windows, but seems more likely that Darwin and linux would handle the same.

Cheers

stellarpower, 2023-01-19:
Also should go without saying, but to confirm, it's creating this in the PWD. So far it had been confined to the installation directory, as that's where I've mostly been running the GUI from, but I've just seen I have this 'file:' directory in my home folder too, from launching the GUI from the desktop launcher in my application menu.

retiutut, 2023-05-18:
I just tried saving BrainFlow CSV to a custom location and it seemed to create a folder and not a file. This should be fixed. Confirmed this issue.