OpenBCI/OpenBCI_GUI #1105: `Error in start_stream:GENERAL_ERROR` from Brainflow when Output to BDF+ (Cyton) and Brainflow streaming is not disabled

stellarpower, 2023-01-16:
## Problem

Opening a live session from a Cyton, with Wifi. 

The data stream is unable to start when:
- The output format for the Cyton recording (older functionality) is BDF+, not "OpenBCI" (ASCII/CSV)
- And the Brainflow streaming to file is enabled (which it is, by default)

Nothing starts in the time series plot, and I get a backtrace to stderr.

On that note - if an exception has been thrown in opening the session/the capture, or some similar error detected, it would probably be good to feed this back to the user with a short message in a modal dialogue, asking them to check the logs using the button.

### BDF+, with Brainflow enabled
```
Console Log Started at Local Time: 2023-01-16_00-51-33
Screen Resolution: 1600 X 900
High-DPI Screen Detected: false
Operating System and Version: Linux - 5.15.0-46-generic
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://docs.openbci.com/Software/OpenBCISoftware/GUIDocs/
OpenBCI_GUI::Setup: GUI v5 Sample Data exists in Documents folder.
Settings: LogFileMaxDuration = 60 minutes
OpenBCI_GUI::Settings: Found and loaded existing GUI-wide Settings from file.
ControlP5 2.3.4 infos, comments, questions at https://github.com/retiutut/controlp5
TopNav: Internet Connection Successful
Local Version: v5.1.0, Latest Version: v5.1.0
GUI is up to date!
BrainFlow Version: fce5573c9456f5216a457fc66171b28a175e8b95
OpenBCI_GUI: AuditoryFeedback: Loading Audio...
OpenBCI_GUI: AuditoryFeedback: Done Loading Audio!
OpenBCI_GUI::Control Panel: Playback history file not found.
OpenBCI_GUI::Setup: Setup is complete!
[ERROR]: OpenBCI_GUI: This application is not being run with Administrator access. This could limit the ability to connect to devices or read/write files.
sourceList: click! 
OpenBCI_GUI: Channel count set to 8
OpenBCI_GUI: Channel count set to 16
[DEFAULT]: Output has been set to BioSemi Data Format (BDF+).
[DEFAULT]: Wifi Devices Refreshing
ControlPanel: User selected Cyton+WiFi 250Hz
wifiList: click! 
[DEFAULT]: Selected WiFi Board: OpenBCI-08E7, WiFi IP Address: 192.168.0.108
[DEFAULT]: Attempting to Start Session...
initButtonPressed: Calling initSystem() after next draw()
OpenBCI_GUI: Start session. Calling initSystem().

=================================================
||             INITIALIZING SYSTEM             ||
=================================================

Sending config string to board: ~6
OpenBCI_GUI: Configuring Cyton Channel Count...
OpenBCI_GUI: Channel count set to 16
"Arial" is not available, so another font will be used. Use PFont.list() to show available fonts.
InitSettings: Saving Default Settings to file!
SessionSettings: 3 active widgets saved!
[SUCCESS]: Session started!
[DEFAULT]: openBCI_GUI: startButton was pressed. Starting data transfer, wait a few seconds.
OpenBCI_GUI: openNewLogFile: opened BDF output file: /home/ben/Documents/OpenBCI_GUI/Recordings/OpenBCI-BDF-2023-01-16_00-52-17.bdf
OpenBCI_GUI: BrainFlow Streamer Location: file:///BrainFlow-RAW_2023-01-16_00-51-34_0.csv:w
Brainflow start streaming
ERROR: Exception when starting stream
brainflow.BrainFlowError: Error in start_stream:GENERAL_ERROR
	at brainflow.BoardShim.start_stream(BoardShim.java:711)
	at OpenBCI_GUI$BoardBrainFlow.startStreaming(OpenBCI_GUI.java:2814)
	at OpenBCI_GUI$BoardCyton.startStreaming(OpenBCI_GUI.java:3619)
	at OpenBCI_GUI.startRunning(OpenBCI_GUI.java:935)
	at OpenBCI_GUI$TopNav.stopButtonWasPressed(OpenBCI_GUI.java:18462)
	at OpenBCI_GUI$TopNav$2.controlEvent(OpenBCI_GUI.java:18301)
	at controlP5.Controller.callListener(Controller.java:1106)
	at controlP5.Controller.setMousePressed(Controller.java:609)
[... 176 lines cut ...]
X11Util: Open[0]: NamedX11Display[:1, 0x7fefa00012c0, refCount 1, unCloseable false]
X11Util: Open[1]: NamedX11Display[:1, 0x7fefa0010ab0, refCount 1, unCloseable false]
```

---

## Operating System and Version

Ubuntu 20.04 AMD64

## GUI Version

Release 5.1.0

## Running standalone app

Standalone

## Type of OpenBCI Board

Cyton+Daisy

## Are you using a WiFi Shield?

Yes

retiutut, 2023-01-18:
Thanks for including enough information to proceed investigating this issue. We have a pretty full lineup for the next version of the GUI that I have started working on, but the fact that this includes WiFi shield unfortunately means this will be lower priority over issues that affect more users.

stellarpower, 2023-01-18:
Thanks - so you're thinking this is tied to hardware? I would've tried MRE with the synthetic board; but unfortunately this doesn't give me the option to save the output data in the same way. I assumed it was in some way to do with trying to output two streams with differing formats at once.

If anyone with a different board could confirm if they can reproduce this, then I guess that would be the next step.

Cheers

closed 2023-09-14 by retiutut