OpenBCI/OpenBCI_GUI #1107: Samples dropped during LSL communication to Matlab

armengol, 2023-01-18:
## Problem

I have a Ganglion board and I'm trying to send data from OpenBCI GUI to Matlab via LSL. The amount of datapoints received during a period of time (ex: 5 seconds) doesn't match with the supposed number of datapoints that I should receive with the given sampling rate (200Hz * 5 seconds = 1000 datapoints, and I get around ~550 datapoints).

If I analyze the timestamps, one every 8 samples is received slower (about 4 times slower).

## Expected 

Receive the amount of samples expected given the sampling rate. If I listen for 5 seconds, I should receive 1000 datapoints. Looking at the timestamps, the difference between two consecutive timestamps should be a constant of 0.005 (so that 1/0.005 = 200 Hz)

## Operating System and Version

macOS 10.15.7

## GUI Version

v5.1.0

## Are you running the downloaded app or from Processing/VSCode?

The downloaded app.

## Type of OpenBCI Board

Ganglion

## Console Log and Screenshots

Paste any relevant text from the console window here, as well as any helpful screenshots of the GUI.
[image]

philippitts, 2023-01-18:
Do you also see this problem when you look at the data recorded by the GUI (vs the data streamed via LSL)? https://docs.openbci.com/Software/OpenBCISoftware/GUIDocs/#access-recorded-data

armengol, 2023-01-18:
No, data is recorded fine by the GUI

retiutut, 2023-01-20:
It's not explicitly mentioned here but are you only talking about TimeSeries output over LSL using Ganglion?

armengol, 2023-01-20:
Yes, I'm talking about TimeSeries output.

I encountered the problem with Matlab, but I just tested your Python script (https://github.com/OpenBCI/OpenBCI_GUI/blob/master/Networking-Test-Kit/LSL/lslStreamTest.py) and the problem persists. One every 8 samples arrives 4 to 10 times late. I attach a screenshot of the output of your Python script. Notice that the script collects data for 5 seconds, and only 512 samples are received.

[image]

retiutut, 2023-01-23:
Awesome, will investigate this asap. Thanks for including all necessary information!

retiutut, 2023-06-13:
It looks like the separate thread which operates the streams was sending out duplicate data much faster than necessary. This is being resolved in #1140.

retiutut, 2023-06-27:
A MATLAB script has now been included in the Networking Test Kit and had passing sampling rates for OpenBCI team.

closed 2023-06-27 by retiutut