OpenBCI/OpenBCI_GUI #1109: GUI will not load past startup

einelson, 2023-01-23:
## Problem

Upon running the GUI for the first time the app will not get past the startup screen. In the screenshot attached below you can see that the app unpackaged the libs and dlls, but nothing beyond that.

## Expected 

I expected after a short wait for the screen to advance to the GUI

## Operating System and Version

Windows 11
Version 22H2
OS build: 22621.1105

## GUI Version

v5.1.0

## Are you running the downloaded app or from Processing/VSCode?

The downloaded app

## Type of OpenBCI Board

Ganglion

## Console Log and Screenshots

[image]

retiutut, 2023-01-30:
Does this computer have an Audio Output specified?

retiutut, 2023-01-31:
Same reason as #1086 but happening on Windows. This is a rare issue.

retiutut, 2023-02-01:
Ok missed this part on first read:

```
So I switched back to windows, plugged (bluetooth-ed) my headphones in and viola! The app started right up.
```

Let's see if we can try/catch this.

retiutut, 2023-02-01:
Replicated on Windows by manually disabling all Audio Devices.

retiutut, 2023-02-01:
[image]

Thankfully the exception is exposed at our level so we are able to catch this! Will then store the fact that Audio Device is not available and output an error when this feature in the Focus widget is pressed.

retiutut, 2023-02-01:
Fun fact. We cannot disable the included audio device on a Mac without using some wizardry. You can remove other third party audio devices, but there is always an included "Built-in Output". Probably for such reasons as this and to allow the system to do beeps and boops.

einelson, 2023-02-01:
Hey! Thanks for the reply! I don't believe my desktop had an audio port specified by default.

Thanks for the quick fix!

PR #1115 (2023-02-02): Fix NullPointerException when no Audio Device is available from the OS
Fixes #1109 #1086
closed 2023-02-08 by retiutut