OpenBCI/OpenBCI_GUI #1111: Upgrade to Processing 4

philippitts, 2023-01-30:
## Problem

Processing 3 is no longer supported and several workarounds are needed in order to continue supporting modern systems. We should upgrade to Processing 4 or enumerate any obstacles that prevent us from doing so.

## Expected 

The application will work when run from Processing 4.

CGrassin, 2023-03-27:
I needed OpenBCI to run on Processing 4.2, so I did a very quick port myself. As far as I can tell, it is extremely straightforward to make it work. The only things I had to fix were a few function calls like the control P5 "setDescription()" method that seems to have been removed.
I can do a PR if that helps, but I did not perform any deep tests, I just used it for my own needs.

retiutut, 2023-06-16:
Comments on Processing 4.2 + Windows:
- There's still a very serious error happening when resizing and trying to close child windows (Console Log and Filter UI pop-outs)
- Until this is fixed we can still move to Processing 4 by making these child windows a fixed height/width and making the non-resizeable

PR #1166 (2023-08-01): Update to Processing 4.2
**Modify App to be fully compatible with Processing 4**

Closes #1111 

There are still issues with child windows not being able to be resized and closed properly. To help us move forward for now, FilterUI, ConsoleLog, and EMG Settings UI have been made non-resizeable. A boolean has been left in FilterUI that would allow this again and easy testing, should a fix become available.

___

**CI/CD Update**

Closes #1170 
Closes #1162 
Closes #1171
closed 2023-08-01 by retiutut