OpenBCI/OpenBCI_GUI #1138: OpenBCI GUI not Loading User-Settings & Auto-deleting User-Settings File

BIRKELANDCOMETH123, 2023-05-27:
## Problem

OpenBCI GUI not Loading User-Settings & Auto-deleting User-Settings File.  

I have determined that the only two JSON settings files that are used on startup of the OpenBCI GUI for my equipment are: “DaisyDefaultSettings.json” and “GuiWideSettings.json”.  In the Settings folder (Documents\OpenBCI_GUI\Settings), these files are created automatically if not present.  In the GuiWideSettings.json file, I have the Expert Mode set to ON as shown in “#1.png”.

See attached two images, “#1.png” and “#2.png”.  See attached log file, “FAILED-USER-SETTINGS_052623_1.txt”.  See attached three JSON settings files, “DaisyDefaultSettings.json”, “GuiWideSettings.json”, and “DESIRED-SETTINGS-052623_1.json”.

[FAILED-USER-SETTINGS_052623_1.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/11580917/FAILED-USER-SETTINGS_052623_1.txt)
[DaisyDefaultSettings.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/11580881/DaisyDefaultSettings.txt)
[DESIRED-SETTINGS-052623_1.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/11580886/DESIRED-SETTINGS-052623_1.txt)
[GuiWideSettings.txt](https://github.com/OpenBCI/OpenBCI_GUI/files/11580887/GuiWideSettings.txt)

The OpenBCI GUI opens fine, the Bluetooth connects to the Cyton-Daisy fine, data is received and displayed and works fine.  I am able to configure the layout the way I want it and save it successfully by clicking “SETTINGS” and then “SAVE”.  This file is attached as “DESIRED-SETTINGS-052623_1.json”. Image #1.png shows what the desired settings’ GUI looks like.
[image]

The image “#2.png” shows what happens when you load the default settings by clicking “SETTINGS” and then “DEFAULT”, and then try to load the user-settings you just saved.  By clicking “SETTINGS” and then “LOAD” and selecting the file “DESIRED-SETTINGS-052623_1.json”, I get a failure notice every time.
[image]

When loading my user-settings file fails, the software ALSO AUTO-DELETES the file that it thought was a problem!  My file “DESIRED-SETTINGS-052623_1.json” gets deleted automatically every time…not to the recycle bin, just gone.  Which is a little bit crazy.

## Expected 

I want to OpenBCI GUI to open to my custom layout every time. 

This problem seems way to BASIC to not be something simple I've missed in the setup.

Please help! Please tell me what I’m doing wrong?  I'm dying to get started.

Thank you!

## Operating System and Version

Windows 10 Pro

## GUI Version

v5.1.0

## Are you running the downloaded app or from Processing/VSCode?

The downloaded app.

## Type of OpenBCI Board

Cyton+Daisy

## Console Log and Screenshots

I've pasted the file(s) above.  But I believe the relevant portion of the log file (attached) is:

SessionSettings: loadConfigFile: User selected C:\Users\XXXXX\Documents\OpenBCI_GUI\OLD-SETTINGS-FILES\DESIRED-SETTINGS-052623_1.json
SessionSettings: Incompatible settings file or other error
[ERROR]: Error trying to load settings file, possibly from previous GUI. Removing old settings.

I put "XXXXX" for the username.

Thank you!

retiutut, 2023-06-12:
Thanks for reporting this bug, this is a coding mistake by myself from a long time ago, which occurs since you are using the last layout. Working on a fix now.

retiutut, 2023-06-12:
Thanks so much for providing the file. The error came from

```
"Current Layout": 12,
``` 

And we mistakenly added 1 when saving to file. There are indeed 12 layouts, but in typical fashion we count from 0, so 12 is out-of-bounds and should be 11. Removed the +1 and it's fixed.

retiutut, 2023-06-12:
@BIRKELANDCOMETH123 As a temporary fix, you could modify this line of your file to be 11 and it will work with 5.1.0. Fix is now present in the `development` branch.

commit bf3372bf (2023-06-12): Fix bug in loading session settings for widget layouts

Fixes #1138
closed 2023-06-12 by retiutut