OpenBCI/OpenBCI_GUI #1141: dylib file not found in jar file

pingwes, 2023-06-08:
## Problem

I'm currently going through this set up guide (https://github.com/OpenBCI/OpenBCI_GUI/wiki/Developer-Setup)

But at the point I run 

`processing-java --force --sketch=/Users/wesberry/prophetic/gui/OpenBCI_GUI --output=/Users/wesberry/prophetic/gui/OpenBCI_GUI/build --run
`

I get the following error:

`file: libGanglionLib.dylib is not found in jar file
file: libneurosdk-shared.dylib is not found in jar file
file: libsimpleble-c.dylib is not found in jar file
file: libMuseLib.dylib is not found in jar file
file: libBrainBitLib.dylib is not found in jar file
file: libBrainFlowBluetooth.dylib is not found in jar file
file: libBoardController.dylib is not found in jar file
OpenBCI_GUI.pde:466:0:466:0: UnsatisfiedLinkError: Unable to load library 'libBoardController.dylib':
dlopen(libBoardController.dylib, 0x0009): tried: '/private/var/folders/7p/71ncf7yn3p5186lrvkqklxb80000gn/T/AppTranslocation/2F1C5EC3-63B0-4DD7-95C2-E5E23C47BACE/d/Processing.app/Contents/PlugIns/jdk-17.0.6+10/Contents/Home/bin/./libBoardController.dylib' (no such file), '/private/var/folders/7p/71ncf7yn3p5186lrvkqklxb80000gn/T/AppTranslocation/2F1C5EC3-63B0-4DD7-95C2-E5E23C47BACE/d/Processing.app/Contents/PlugIns/jdk-17.0.6+10/Contents/Home/bin/../lib/libBoardController.dylib' (no such file), 'libBoardController.dylib' (no such file), '/usr/lib/libBoardController.dylib' (no such file), '/private/var/folders/7p/71ncf7yn3p5186lrvkqklxb80000gn/T/AppTranslocation/2F1C5EC3-63B0-4DD7-95C2-E5E23C47BACE/d/Processing.app/Contents/Java/libBoardController.dylib' (no such file)`

## Operating System and Version

macOS/Windows/Linux - Example: MacOS 12.3.1

## GUI Version

5.1.0

## Are you running the downloaded app or from Processing/VSCode?

Running the code.

## Console Log and Screenshots

` processing-java --force --sketch=/Users/wesberry/prophetic/gui/OpenBCI_GUI --output=/Users/wesberry/prophetic/gui/OpenBCI_GUI/build --run
Console Log Started at Local Time: 2023-06-08_15-29-51
Screen Resolution: 1440 X 900
High-DPI Screen Detected: true
Operating System and Version: Mac - 12.3.1
MacOS Details: MacOS Monterey
Welcome to the Processing-based OpenBCI GUI!
For more information, please visit: https://docs.openbci.com/Software/OpenBCISoftware/GUIDocs/
OpenBCI_GUI::Setup: GUI v5 Sample Data exists in Documents folder.
Settings: LogFileMaxDuration = 60 minutes
OpenBCI_GUI::Settings: Found and loaded existing GUI-wide Settings from file.
ControlP5 2.3.4 infos, comments, questions at https://github.com/retiutut/controlp5
TopNav: Internet Connection Successful
Local Version: v5.1.0, Latest Version: v5.1.0
GUI is up to date!
file: libGanglionLib.dylib is not found in jar file
file: libneurosdk-shared.dylib is not found in jar file
file: libsimpleble-c.dylib is not found in jar file
file: libMuseLib.dylib is not found in jar file
file: libBrainBitLib.dylib is not found in jar file
file: libBrainFlowBluetooth.dylib is not found in jar file
file: libBoardController.dylib is not found in jar file
OpenBCI_GUI.pde:466:0:466:0: UnsatisfiedLinkError: Unable to load library 'libBoardController.dylib':
dlopen(libBoardController.dylib, 0x0009): tried: '/private/var/folders/7p/71ncf7yn3p5186lrvkqklxb80000gn/T/AppTranslocation/2F1C5EC3-63B0-4DD7-95C2-E5E23C47BACE/d/Processing.app/Contents/PlugIns/jdk-17.0.6+10/Contents/Home/bin/./libBoardController.dylib' (no such file), '/private/var/folders/7p/71ncf7yn3p5186lrvkqklxb80000gn/T/AppTranslocation/2F1C5EC3-63B0-4DD7-95C2-E5E23C47BACE/d/Processing.app/Contents/PlugIns/jdk-17.0.6+10/Contents/Home/bin/../lib/libBoardController.dylib' (no such file), 'libBoardController.dylib' (no such file), '/usr/lib/libBoardController.dylib' (no such file), '/private/var/folders/7p/71ncf7yn3p5186lrvkqklxb80000gn/T/AppTranslocation/2F1C5EC3-63B0-4DD7-95C2-E5E23C47BACE/d/Processing.app/Contents/Java/libBoardController.dylib' (no such file)
dlopen(libBoardController.dylib, 0x0009): tried: '/private/var/folders/7p/71ncf7yn3p5186lrvkqklxb80000gn/T/AppTranslocation/2F1C5EC3-63B0-4DD7-95C2-E5E23C47BACE/d/Processing.app/Contents/PlugIns/jdk-17.0.6+10/Contents/Home/bin/./libBoardController.dylib' (no such file), '/private/var/folders/7p/71ncf7yn3p5186lrvkqklxb80000gn/T/AppTranslocation/2F1C5EC3-63B0-4DD7-95C2-E5E23C47BACE/d/Processing.app/Contents/PlugIns/jdk-17.0.6+10/Contents/Home/bin/../lib/libBoardController.dylib' (no such file), 'libBoardController.dylib' (no such file), '/usr/lib/libBoardController.dylib' (no such file), '/private/var/folders/7p/71ncf7yn3p5186lrvkqklxb80000gn/T/AppTranslocation/2F1C5EC3-63B0-4DD7-95C2-E5E23C47BACE/d/Processing.app/Contents/Java/libBoardController.dylib' (no such file)
Native library (darwin/libBoardController.dylib) not found in resource path (/Users/wesberry/prophetic/gui/OpenBCI_GUI/build::/Users/wesberry/prophetic/gui/OpenBCI_GUI/code/LSLLink.jar:/private/var/folders/7p/71ncf7yn3p5186lrvkqklxb80000gn/T/AppTranslocation/2F1C5EC3-63B0-4DD7-95C2-E5E23C47BACE/d/Processing.app/Contents/Java/core/library/jogl-all.jar:/private/var/folders/7p/71ncf7yn3p5186lrvkqklxb80000gn/T/AppTranslocation/2F1C5EC3-63B0-4DD7-95C2-E5E23C47BACE/d/Processing.app/Contents/Java/core/library/gluegen-rt.jar:/private/var/folders/7p/71ncf7yn3p5186lrvkqklxb80000gn/T/AppTranslocation/2F1C5EC3-63B0-4DD7-95C2-E5E23C47BACE/d/Processing.app/Contents/Java/core/library/core.jar:/Users/wesberry/Documents/Processing/libraries/LSLLink/library/LSLLink.jar:/Users/wesberry/Documents/Processing/libraries/LSLLink/library/jna.jar:/Users/wesberry/Documents/Processing/libraries/LSLLink/library/jna-platform.jar:/Users/wesberry/Documents/Processing/libraries/minim/library/tritonus_share.jar:/Users/wesberry/Documents/Processing/libraries/minim/library/mp3spi1.9.5.jar:/Users/wesberry/Documents/Processing/libraries/minim/library/jl1.0.1.jar:/Users/wesberry/Documents/Processing/libraries/minim/library/minim.jar:/Users/wesberry/Documents/Processing/libraries/minim/library/jsminim.jar:/Users/wesberry/Documents/Processing/libraries/minim/library/tritonus_aos.jar:/private/var/folders/7p/71ncf7yn3p5186lrvkqklxb80000gn/T/AppTranslocation/2F1C5EC3-63B0-4DD7-95C2-E5E23C47BACE/d/Processing.app/Contents/Java/modes/java/libraries/serial/library/jssc.jar:/private/var/folders/7p/71ncf7yn3p5186lrvkqklxb80000gn/T/AppTranslocation/2F1C5EC3-63B0-4DD7-95C2-E5E23C47BACE/d/Processing.app/Contents/Java/modes/java/libraries/serial/library/serial.jar:/private/var/folders/7p/71ncf7yn3p5186lrvkqklxb80000gn/T/AppTranslocation/2F1C5EC3-63B0-4DD7-95C2-E5E23C47BACE/d/Processing.app/Contents/Java/modes/java/libraries/net/library/net.jar:/Users/wesberry/Documents/Processing/libraries/grafica/library/grafica.jar:/Users/wesberry/Documents/Processing/libraries/GifAnimation/library/GifAnimation.jar:/Users/wesberry/Documents/Processing/libraries/oscP5/library/oscP5.jar:/Users/wesberry/Documents/Processing/libraries/udp/library/udp.jar:/Users/wesberry/Documents/Processing/libraries/jSerialComm/library/jSerialComm.jar:/Users/wesberry/Documents/Processing/libraries/brainflow/library/brainflow.jar:/Users/wesberry/Documents/Processing/libraries/httprequests_processing/library/commons-logging-1.1.1.jar:/Users/wesberry/Documents/Processing/libraries/httprequests_processing/library/httpclient-4.1.2.jar:/Users/wesberry/Documents/Processing/libraries/httprequests_processing/library/httpcore-4.1.2.jar:/Users/wesberry/Documents/Processing/libraries/httprequests_processing/library/commons-codec-1.4.jar:/Users/wesberry/Documents/Processing/libraries/httprequests_processing/library/httpclient-cache-4.1.2.jar:/Users/wesberry/Documents/Processing/libraries/httprequests_processing/library/httprequests_processing.jar:/Users/wesberry/Documents/Processing/libraries/httprequests_processing/library/httpmime-4.1.2.jar:/Users/wesberry/Documents/Processing/libraries/controlP5/library/controlP5.jar:/Users/wesberry/Documents/Processing/libraries/openbci_gui_helpers/library/openbci_gui_helpers.jar:/Users/wesberry/Documents/Processing/libraries/ssdp_client/library/ssdp_client.jar)
A library used by this sketch relies on native code that is not available.
UnsatisfiedLinkError: Unable to load library 'libBoardController.dylib':
dlopen(libBoardController.dylib, 0x0009): tried: '/private/var/folders/7p/71ncf7yn3p5186lrvkqklxb80000gn/T/AppTranslocation/2F1C5EC3-63B0-4DD7-95C2-E5E23C47BACE/d/Processing.app/Contents/PlugIns/jdk-17.0.6+10/Contents/Home/bin/./libBoardController.dylib' (no such file), '/private/var/folders/7p/71ncf7yn3p5186lrvkqklxb80000g [... cut ...] ry/httpclient-cache-4.1.2.jar:/Users/wesberry/Documents/Processing/libraries/httprequests_processing/library/httprequests_processing.jar:/Users/wesberry/Documents/Processing/libraries/httprequests_processing/library/httpmime-4.1.2.jar:/Users/wesberry/Documents/Processing/libraries/controlP5/library/controlP5.jar:/Users/wesberry/Documents/Processing/libraries/openbci_gui_helpers/library/openbci_gui_helpers.jar:/Users/wesberry/Documents/Processing/libraries/ssdp_client/library/ssdp_client.jar)
≈
`

retiutut, 2023-06-10:
The files are there, but you were unable to unpack them. I'm not sure why this is the case, but we have done development on MacOS 12.

closed 2023-06-10 by retiutut

pingwes, 2023-06-22:
So it was a macOS issue. OS wouldn't allow program to access files in processing directory for some reason. After I updated OS I'm able to access files