OpenBCI/OpenBCI_GUI #1146: Add relevant help tooltips or docs links within widgets

jartuso, 2023-06-22:
## **Is your feature request related to a problem? Please describe.**
Brand new users of the GUI may not know where to go to find documentation. People can be pretty lazy or may not think to go read the docs. Adding tooltips could help reduce the amount of support tickets from confused 1st time users.

For example --  The NeuroFly EMG Joystick comes with an EMG Settings widget. As a 1st time user, I didn't know what the various EMG settings did and I didn't know (forgot) that it was already well-documented [here](https://docs.openbci.com/Software/OpenBCISoftware/NeuroFly_Toolkit/)

## **Describe the solution you'd like**
Add a (?) icon to each widget that includes a link to the relevant section of the OpenBCI docs. 

## **Describe alternatives you've considered**
A single catch-all documentation link.
Adding tooltip text that explain things within the GUI, rather than linking out to do the docs

## **Additional context**

bobthefairycat, 2023-06-24:
If this is going to be implemented, I think users should be given an option at first startup or the like to enable or disable tooltips. I personally abhor tooltips.

retiutut, 2023-06-27:
Great feedback! Prioritizing this. I also like the idea to somehow hide what would be buttons for experienced users.