OpenBCI/OpenBCI_GUI #1172: Add check for workflow dispatch to deploy builds to AWS S3

retiutut, 2023-07-31:
Towards the end of this sprint/cycle, we can re-add on-workflow dispatch to deploying builds to S3 for internal testing.

Example:
```
- name: Store Build on AWS
      if: github.event_name == 'workflow_dispatch'
      run: |
        cd $GITHUB_WORKSPACE
        ls
        export CURRENT_BRANCH=`git branch --show-current`
        export GUI_COMMIT_TIME=`cat temp/timestamp.txt`
        export GUI_VERSION_STRING=`cat temp/versionstring.txt`
        echo $GUI_COMMIT_TIME
        echo $GUI_VERSION_STRING
        echo $CURRENT_BRANCH
        aws s3 rm s3://openbci-public-gui-v6/latest  --recursive --exclude "*" --include "openbcigui_*_macosx.dmg"
        aws s3 cp $GITHUB_WORKSPACE/. s3://openbci-public-gui-v6/${GUI_VERSION_STRING}  --recursive --exclude "*" --include "openbcigui_*_macosx.dmg"
        aws s3 cp $GITHUB_WORKSPACE/. s3://openbci-public-gui-v6/latest  --recursive --exclude "*" --include "openbcigui_*_macosx.dmg"
```

This will also help us limit the amount of times we use Windows signing certificate.

We need these Github Actions ymls pushed to `master` before they even become available.

philippitts, 2023-09-18:
I think this is no longer needed. We have workflows now that build artifacts for active PRs, so there shouldn't be a need for manual dispatching. We also don't have limitations on the number of times we use the Windows signing certificate because we elected to go with a different strategy for that certificate.

retiutut, 2023-09-18:
Sweet, thanks for chiming in about this. Closing this issue.

closed 2023-09-18 by retiutut