OpenBCI/OpenBCI_GUI #1204: Impedance checks Cyton incorrect at low resistances

philippitts, 2023-11-03:
## Problem

Several users on the [OpenBCI forum](https://openbci.com/forum/index.php?p=/discussion/3726/electrode-impedance-test-using-the-gui-widget) report incorrect values reported by the electrode impedance widget for Cyton when using resistors of 500kOhms or less. Using resistors of greater than 500kOhms appear to be working as expected.

## Expected 

The Cyton impedance check widget should reflect the values of resistors used in lab validation configurations. If a resistor is connected between the N port of a channel and SRB2, the widget should show an impedance similar to the value of the resistor.

philippitts, 2023-11-05:
Below are a series of tests surrounding the problem space. I connected several different resistors between the negative channel 1 pin and each of the following pins: SRB1, SRB2, Channel 1 positive, BIAS 1, BIAS 2, AGND 1, and AGND 2. I performed these tests with a new Cyton using the default settings (including hardware settings) in the GUI. I used GUI version 5.2.2 with Windows 10.

Below are my setups for each of these configurations.

**SRB1**

[image]

**SRB2**

[image]

**Channel 1 Positive**

[image]

**BIAS1**

[image]

**BIAS2**

[image]

**AGND1**

[image]

**AGND2**

[image]

---

The table below shows the values returned in the OpenBCI GUI Cyton Signal widget for each of the different setups. I waited 5 seconds before recording each test for the value to settle.

| Resistor Value | Pin | GUI Impedance Reading |
| --------------- | --- | --------------------------- |
| 1000 kOhm | SRB1 | 1013 kOhm |
| 1000 kOhm | SRB2 | 0 kOhm |
| 1000 kOhm | 1P | 522 kOhm |
| 1000 kOhm | BIAS1 | ~400 kOhm (dropping) |
| 1000 kOhm | BIAS2 | ~250 kOhm (dropping) |
| 1000 kOhm | AGND1 | 973 kOhm |
| 1000 kOhm | AGND2 | 973 kOhm |

| Resistor Value | Pin | GUI Impedance Reading |
| --------------- | --- | --------------------------- |
| 680 kOhm | SRB1 | 0 kOhm |
| 680 kOhm | SRB2 | 825 kOhm |
| 680 kOhm | 1P | 373 kOhm |
| 680 kOhm | BIAS1 | ~300 kOhm (dropping) |
| 680 kOhm | BIAS2 | ~150 kOhm (dropping) |
| 680 kOhm | AGND1 | 715 kOhm |
| 680 kOhm | AGND2 | 715 kOhm |

| Resistor Value | Pin | GUI Impedance Reading |
| --------------- | --- | --------------------------- |
| 470 kOhm | SRB1 | 2510 kOhm |
| 470 kOhm | SRB2 | 0 kOhm |
| 470 kOhm | 1P | 256 kOhm |
| 470 kOhm | BIAS1 | 30 kOhm  (dropping) |
| 470 kOhm | BIAS2 | 4 kOhm (stable) |
| 470 kOhm | AGND1 | 479 kOhm |
| 470 kOhm | AGND2 | 479 kOhm |

| Resistor Value | Pin | GUI Impedance Reading |
| --------------- | --- | --------------------------- |
| 51 kOhm | SRB1 | 2450 kOhm |
| 51 kOhm | SRB2 | 0 kOhm |
| 51 kOhm | 1P | 28 kOhm |
| 51 kOhm | BIAS1 | ~80 kOhm (dropping)
| 51 kOhm | BIAS2 | 15 kOhm (dropping)
| 51 kOhm | AGND1 | 53 kOhm |
| 51 kOhm | AGND2 | 55 kOhm |

---

General observations:
* The SRB pins were unstable and would either show very high impedance or no impedance (0 kOhm).
* The positive channel pin appeared to show about half the correct impedance.
* The BIAS pins were unstable and would consistently report lower values over time.
* The AGND pins reported imperfect, but generally accurate results.

wjcroft, 2023-11-05:
Phil, thanks!

Seems pretty certain that SRB2 should be the pin used, in normal EEG applications. And that SRB2 PREVIOUSLY worked with the original ADS1299 register settings. Which is confirmed on this page,

https://openbci-stream.readthedocs.io/en/latest/notebooks/A2-electrodes_impedance.html

So hopefully this is an easy fix.

Thanks everyone.

retiutut, 2024-02-27:
https://openbci.com/forum/index.php?p=/discussion/3790/high-impedance#latest

philippitts, 2024-02-29:
Additional testing with different versions of the GUI:

GUI v5.0.0 👍 
| Resistor Value | Pin | GUI Impedance Reading |
| --------------- | --- | --------------------------- |
| 51 kOhm | SRB1 | 197 kOhm |
| 51 kOhm | SRB2 | 48.9 kOhm |
| 51 kOhm | 1P | 48.9 kOhm |
| 51 kOhm | BIAS1 | ~40 kOhm (unstable) |
| 51 kOhm | BIAS2 | 0 kOhm |
| 51 kOhm | AGND1 | 0 kOhm |
| 51 kOhm | AGND2 | 0 kOhm |

GUI v5.0.6 👍 
| Resistor Value | Pin | GUI Impedance Reading |
| --------------- | --- | --------------------------- |
| 51 kOhm | SRB1 | 0 kOhm |
| 51 kOhm | SRB2 | 48.8 kOhm |
| 51 kOhm | 1P | 49 kOhm |
| 51 kOhm | BIAS1 | 0 kOhm |
| 51 kOhm | BIAS2 | 0 kOhm |
| 51 kOhm | AGND1 | 0 kOhm |
| 51 kOhm | AGND2 | 0 kOhm |

GUI v5.0.8 👍 
| Resistor Value | Pin | GUI Impedance Reading |
| --------------- | --- | --------------------------- |
| 51 kOhm | SRB1 | 0 kOhm |
| 51 kOhm | SRB2 | 48.8 kOhm |
| 51 kOhm | 1P | 49 kOhm |
| 51 kOhm | BIAS1 | 0 kOhm |
| 51 kOhm | BIAS2 | 0 kOhm |
| 51 kOhm | AGND1 | 0 kOhm |
| 51 kOhm | AGND2 | 0 kOhm |

GUI v5.0.9 👎 
| Resistor Value | Pin | GUI Impedance Reading |
| --------------- | --- | --------------------------- |
| 51 kOhm | SRB1 | 0 kOhm |
| 51 kOhm | SRB2 | 511 kOhm |
| 51 kOhm | 1P | 27 kOhm |
| 51 kOhm | BIAS1 | ~16 kOhm |
| 51 kOhm | BIAS2 | ~21 kOhm |
| 51 kOhm | AGND1 | ~62 kOhm (unstable) |
| 51 kOhm | AGND2 | ~64 kOhm (unstable) |

GUI v5.1.0 👎  
| Resistor Value | Pin | GUI Impedance Reading |
| --------------- | --- | --------------------------- |
| 51 kOhm | SRB1 | ~2400 kOhm |
| 51 kOhm | SRB2 | 0 kOhm |
| 51 kOhm | 1P | 57 kOhm |
| 51 kOhm | BIAS1 | ~30 kOhm |
| 51 kOhm | BIAS2 | ~2 kOhm |
| 51 kOhm | AGND1 | ~53 kOhm |
| 51 kOhm | AGND2 | ~53 kOhm |

GUI v5.2.0 👎  
| Resistor Value | Pin | GUI Impedance Reading |
| --------------- | --- | --------------------------- |
| 51 kOhm | SRB1 | 516 kOhm |
| 51 kOhm | SRB2 | 2450 kOhm |
| 51 kOhm | 1P | 27.7 kOhm |
| 51 kOhm | BIAS1 | ~75 kOhm |
| 51 kOhm | BIAS2 | ~75 kOhm |
| 51 kOhm | AGND1 | ~70 kOhm |
| 51 kOhm | AGND2 | ~70 kOhm |

The problem appears to have been introduced in v5.0.9

closed 2025-05-01 by retiutut