OpenBCI/OpenBCI_GUI #1228: Download filtered data directly from the GUI

gabrielacurielr, 2025-01-29:
## **Is your feature request related to a problem? Please describe.**
I purchased the headset to get the information from experiments without the necessity to know about programming. I´m frustrated when I record an experiment and want to export the data I can visualize in FFT Plot and Band Power Widget, but only get the raw data per electrode; I asked in the forum and via mail but the answer is to redirect to other platforms,

## **Describe the solution you'd like**
An option to export the data behind or resulting from the widgets, such as the FFT Plot an Band Power.

## **Describe alternatives you've considered**
A feature that allows to export the filtered data that´s being shown through the graphs.

## **Additional context**
Not needed.

CatherineFeijoo, 2025-01-30:
Yes! I´m looking forward to an option like this. I purchased the headset as it was promoted to be useful even if you didn't know programming, however the solution I got in the forum and via mail is to send the data to other platforms with coding involved. I´d love to get access to the filtered data that's shown in graphs.

retiutut, 2025-03-31:
The Networking Widget allows you to export the data in real time. You could then receive this data and save it to file using any method you choose.

retiutut, 2025-03-31:
Given the initial information from you both, I've renamed the ticket to `Add ability to save FFT and Band Power data to file`. Does this new name seem to resolve your concerns?