OpenBCI/OpenBCI_GUI #1238: railed channel in cyton+daisy using EEG- MarkIV headset 16 channels

shanullah, 2025-02-27:
## Problem

Explain the problem.

## Expected 

In a perfect world, what do you expect to happen.

## Operating System and Version

macOS/Windows/Linux - Example: MacOS 10.15.7

## GUI Version

The version is displayed on startup and at the top of every Console Log.

## Are you running the downloaded app or from Processing/VSCode?

The downloaded app or compiling and running the code.

## Type of OpenBCI Board

Cyton/Cyton+Daisy/Ganglion

## Console Log and Screenshots

Paste any relevant text from the console window here, as well as any helpful screenshots of the GUI.

shanullah, 2025-02-27:
We have been facing issue with particular channels , 12 and 15, no matter how much contact with scalp we make it shows continuously railed, however in the beginning of gui, all channels shows stream/start signals.
can you suggest what's going on here. 
one thing we observed is that our particular electrodes might have issue like the screw is very tight, and we are unable to change or replace the tip of electrodes while for others we could do, but apparently it seems fine. do we have any others for its functionality

shanullah, 2025-02-27:
[image]

retiutut, 2025-03-31:
@shanullah Please reach out to official OpenBCI support, if you have not already.

closed 2025-03-31 by retiutut